pybricks/support #367: how to read from a keyboard?

henry333, 2021-06-08:
have connected a keyboard to ev3 via usb, can read from the keyboard in c or python3, simply like:
message=input()
and it can work.
but not successful in pybricks-micropython, does it work at this moment?

laurensvalk, 2021-06-08:
For inspiration, check out [this example for a PS4 controller](https://github.com/pybricks/pybricks-api/blob/master/examples/ev3/ps4/main.py). This is a nice example of how input devices can be read without the `evdev` library, by essentially doing everything yourself. And then you'd want to do something like it but for the keyboard, perhaps by making a MicroPython version of this [C example](https://stackoverflow.com/a/20946151).

So, this could be a bit of a puzzle, but it's probably not impossible.

henry333, 2021-06-08:
didn't realize it could be this complicated, simply thought to use a keyboard as a standard input, just like this:
#! /usr/bin/env pybricks-micropython
from pybricks.hub import EV3Brick
ev3=EV3Brick()
message=input()
ev3.screen.clear(0
ev3.screen.print(message)

the display showed the large arrow then froze, keyboard or buttons wouldn't response, ctrl+atl+F either. 
The program can work via SSH, read from laptop's keyboard and showed the message on EV3's display. I thought it must be something with the pybricks-micropython handling  input..

henry333, 2021-06-08:
c or python3 can read the keyboard from USB port on EV3 Brick, must find some way in pybricks-mpy, still thanks a lot! @laurensvalk

JorgePe, 2021-06-08:
> Ah -- the (complicated) examples above are useful if you want to connect the keyboard to the USB port on the EV3 Brick itself.
> 
> Yes, reading your computer's keyboard might be easier. I don't know if `input` currently works in `pybricks-micropython`. I don't remember trying this.

input() works (if you start the script wth 'brickrun -r -- pybricks-micropython test_kbd-m.py'

```
#!/usr/bin/env pybricks-micropython

while True:
    print(input())
```

but input requires ENTER so no fast gaming this way ;)

dlech, 2021-06-08:
On the EV3, Pybricks MicroPython runs in graphics mode, so the keyboard input is taken over from the (text) console. This is why `input()` isn't working. What sort of input are you trying to get?

henry333, 2021-06-09:
like to use a keyboard to send command to spare a tablet or laptop. Thanks to @laurensvalk the example is very inspiring. 
Here is a short script works now:
#! /usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
import struct
infile_path="/dev/input/event2"
in_file=open(infile_path,"rb")
FORMAT='llHHi'
EVENT_SIZE=struct.calcsize(FORMAT)
event=in_file.read(EVENT_SIZE)
ev3=EV3Brick();
ev3.screen.clear();
while event:
	(tv_sec,tv_usec,ev_type,code,value)=struct.unpack(FORMAT,event);
	if 1==ev_type:
		if 17==code and 1==value:
			ev3.screen.print("W is pressed");
		if (1==code) and 0==value:
			break;
	event=in_file.read(EVENT_SIZE);
in_file.close();

will try to synchronize the keyboard with all other movements. And to @JorgePe, your method can work via SSH, thanks bro!

closed 2021-06-11 by henry333