pybricks/support #368: [Feature] directly connect two technichub and communicate with dual-way input()/print()

ruf, 2021-06-10:
The move/city/technichub inited a basic bluetooth lib and waiting for PC connection. When connected, the stdin/stdout works, so we can use input()/print() in micropython, to communicate with PC.

What if one of the hub use current bluetooth lib to connect another hub, and create a pipe between two hubs? One's print() is another's input(). 

For example, 3 new methods:

hub.disconnect_console()    # disconnect and waiting for new incoming connection
hub.connect_console(mac_address)   # connect another hub
hub.is_connected()           # return mac_address if connected, or None if not connected.

If it works, we can create bigger projects with 2 hubs and 8 ports, more powerful than primehub!?

And ... if we can create multiple NUS BT pipes between hubs, maybe it is a easy way to cooperated with many hubs?

dlech, 2021-06-10:
Hi, we have a discussion about this already in #262.

ruf, 2021-06-10:
Full functional pybricks.messaging should be better. But it need more designing and coding, and more memory (maybe).

I just brainstorming what that absolute minimum might be... maybe a little tweak in pbsys_bluetooth_process()? one hub is advertising and another is connecting ...? when the connection is established, there is no more works to do, just use stdin/stdout as usual.

just brainstorming :)

laurensvalk, 2021-06-10:
> one hub is advertising and another is connecting ...? when the connection is established, there is no more works to do, just use stdin/stdout as usual.

We've considered this. But the problem is that the IDE also uses stdin/stout, so you couldn't program the hub while it is connected to another hub. So that's why scenario **3a** in https://github.com/pybricks/support/issues/262 is being considered. Hub-to-hub communication is kept separate from download-and-run.

Wireless communication can be challenging for new programmers, so that's why we're aiming to keep things simple. 

That said, you're welcome to dig into the source code and try that "tweak" you mentioned. We're definitely curious to see what you'll make :wink: