pybricks/support #375: [Feature] Add letter beta to app icon

laurensvalk, 2021-06-25:
**Is your feature request related to a problem? Please describe.**

It would be nice to make the Pybricks Code icon slightly different for the beta version so it's easy to see which one you use.

Perhaps we could add the letter β in a corner of the logo.

This is especially useful in cases where the full name is not shown.

[image]

dlech, 2021-06-25:
How about just adding a badge to the icon? 
[image]

We could also just make the name "Pybricks Beta" so it doesn't get cut off so easily.

dlech, 2021-06-25:
We should also have a maskable version of the icon as described [here](https://web.dev/maskable-icon/).

laurensvalk, 2021-06-25:
A badge is good; anything to easily tell them apart will do.

closed 2021-07-08 by dlech