pybricks/support #376: [Question] Does  pybricks.experimental import getchar  still exist?

johnscary-ev3, 2021-06-30:
Does "from pybricks.experimental import getchar" still work?

Updated my Robot Inventor Hub using https://install.pybricks.com/

The getchar function does not seem to exist now but I would like to use it for Remote control.

```
13 from pybricks.experimental import getchar

File "robot.py", line 13, in <module>
ImportError: can't import name getchar
```

Do I need a special version?
I don't see getchar in the documentation either.

Thanks.

laurensvalk, 2021-06-30:
You can now read from `stdin` to achieve the same result. We have an example project [here](https://pybricks.com/projects/sets/technic/42099-off-roader/keyboard-remote/). Now that this feature is available, we have deprecated `getchar`.

By the way, we recently added experimental support for the Powered Up remote, which may be useful as well. See #186.

johnscary-ev3, 2021-06-30:
Hi,
Thanks for the input. 
I changed the getchar calls over to stdin calls as in the example.
No Python import errors now, but I have another problem
I am using the pybricksdev software on the PC side to talk to Robot Inventor.
That software does the BLE connection and downloads my program fine from Visual Studio Code.
The Inventor program starts and is running ok, sending messages and doing actions.
However when PC side goes to check status of program running on hub, it never sees the hub running now.
That is, it never gets out of this loop below. Any inputs on that?

```
# Give the hubs some time to start
    while hub.state != hub.RUNNING :
        print('forwarder: Waiting for Hub to Run')
        await sleep(2)
    print('forwarder: Hub Running')
```
PS: I also noticed that some of the Color Sensor handling changed, but I worked my way through that using the new docs ;0)

dlech, 2021-06-30:
What is `pybricksdev --version`?

johnscary-ev3, 2021-06-30:
Looks like version = "0.1.0"
Do I need an update?
Thanks.

```
[tool.poetry]
name = "pybricksdev-demo"
version = "0.1.0"
description = "Interact with multiple Pybricks hubs in a Jupyter-Notebook using pybricksdev"
authors = ["The Pybricks Authors <dev@pybricks.com>"]
maintainers = ["Laurens Valk <laurens@pybricks.com>"]
license = "MIT"
readme = "README.md"
homepage = "https://pybricks.com"
repository = "https://github.com/pybricks/pybricksdev-demo"
documentation = "https://docs.pybricks.com"
```

dlech, 2021-06-30:
I don't think `pybricksdev-demo` has been kept up-to-date with recent changes to `pybricksdev`. Are you writing your code in a notebook?

johnscary-ev3, 2021-06-30:
Actually I started off using the notebook RC demo but then converted it to a normal Python program in the same project folder with the notebook stuff.
Should I change over to just pybricksdev project and move my Python programs over to that to get latest updates?

dlech, 2021-06-30:
If you just want to download and run a program, you can use `pybricksdev` as a command line tool (although we would need to fix https://github.com/pybricks/pybricksdev/issues/17 first for your remote control to work).

I guess I would need to see the whole project folder to be able to tell you what needs to be done to fix it.

johnscary-ev3, 2021-06-30:
Ok well it looks like current situation is that the pybricksdev package can do a hub.run command ok but after that we can just read outputs from the hub but not send any commands to it. Since I updated the hub and not the pybricksdev on my PC, I would assume the changes to the hub software is causing this, or pybricksdev needs a fix to work with updated hub sw as you mention in pybrickdev#17 above. Command line mode does not really help with getting the RC thing going.
Looks like updating my pybrickdev package would not help either since there is a known issue (#17) with that now.

I suppose I could try using the browser based environment which I assume would work using the example given above by Laurens. Copy my Hub program over and run it in that environment for now?
Thanks.

dlech, 2021-06-30:
The issue I referenced only applies to the command line mode, but yes, we recommend using https://code.pybricks.com or https://beta.pybricks.com.

johnscary-ev3, 2021-07-01:
Ok I am using code.pybricks.com and also the app program.
These work great with my program as far as doing character based RC commands ;0)
Still would be nice to be able to use VSC as before since then we have git, file management, and a powerful editor, etc.
By the way, would be nice to have a "save as" command in the pybricks environment, but that is another "issue/request" I guess.
Thanks much.
 Appreciate the support and all your work on pybricks!

dlech, 2021-07-01:
Currently, you have to change the "Ask where to save each file before downloading" setting in your browser to get "save as".

laurensvalk, 2021-07-01:
> Ok well it looks like current situation is that the pybricksdev package can do a hub.run command ok but after that we can just read outputs from the hub but not send any commands to it.

It looks like we removed the `write` method to send data.  We can restore that in a future release.

In the mean time, you can do it yourself like this:

```python
from pybricksdev.ble import nus
await hub.client.write_gatt_char(nus.NUS_RX_UUID, bytearray(b"HELLO!"), True)
```

Here is a minimal working example:

[image]

laurensvalk, 2021-07-01:
This is expected to change eventually though, when we enable hub-to-hub communication. Then, your PC will be able to communicate to the hub in the same way, and independent from the whole download-and-run process, making it all a bit more reliable.

johnscary-ev3, 2021-07-01:
@laurensvalk 
Thanks so much for the help on this.
I think I see how to use this way to replace the hub.write method.
However I am wondering about the hub.status parameter that we used before to check on the hub to make sure it was running before doing any hub.write actions.
Do we not need to use that now? Don't see it in your example above.
Currently it is not not reporting hub.RUNNING so it just hangs in this checking loop even though hub program is clearly running and sending me messages and taking actions.

```
# Give the hubs some time to start
    while hub.state != hub.RUNNING :
        print('forwarder: Waiting for Hub to Run')
        await sleep(2)
    print('forwarder: Hub Running')
```
PS: I am looking forward to new hub-to-hub methods which should open up a lot of interesting possibilities.

laurensvalk, 2021-07-01:
Indeed, you may not need to check for the run state anymore. But if you find that you still need it, how about putting something like this as the first line of your program?

print(“started!”)

and then check for that output before you send anything important to the hub.

dlech, 2021-07-01:
there is now a boolean `program_running` attribute that can be used for this.

johnscary-ev3, 2021-07-01:
Hi guys,
After some diagnostic print statements, I found that hub,state now goes through this sequence  0, 4, 4,...,1,1, ....
The 1 is when program starts running. 
However hub.RUNNING is equal to 2, so my checking loop gets stuck. Changed it to check for ==1 and it now works.
Maybe my version of pybricksdev code is old and this is fixed?  I tried hub.program_running but did not work, maybe also due to old code I have?

johnscary-ev3, 2021-07-02:
Sorry to bother you all again.
I did a "poetry update" to get latest pybricksdev code.
First deleted .venv and poetry reloaded it.
Then I tried to use your example code above in my Python program,
Now I just get a timeout when it is trying to find the device.
The normal pybricks environment finds it ok.
This is main() routine I have for finding and connecting.
Any ideas?

```
from pybricksdev.connections import PybricksHub
from pybricksdev.ble import find_device
from asyncio import gather, sleep, run
import msvcrt

hub_RUNNING = 1

async def main():
    print('main: Start')
    hub = PybricksHub()   
    # You can search for the address like this:
    address = await find_device('Pybricks Hub')
    await hub.connect(address)
    await gather(
    hub.run('robot.py', print_output=True),
    forwarder(hub)
    )

    # Disconnect from the hub
    await hub.disconnect()
    print('main: Stop')
```

```
PS C:\Users\johns\pybricksdev-demo> & C:/Users/johns/pybricksdev-demo/.venv/Scripts/python.exe c:/Users/johns/pybricksdev-demo/RemoteControl.py
main: Start
Traceback (most recent call last):
  File "c:/Users/johns/pybricksdev-demo/RemoteControl.py", line 60, in <module>
    run(main())
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.2800.0_x64__qbz5n2kfra8p0\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.2800.0_x64__qbz5n2kfra8p0\lib\asyncio\base_events.py", line 616, in run_until_complete    
    return future.result()
  File "c:/Users/johns/pybricksdev-demo/RemoteControl.py", line 12, in main
    address = await find_device('Pybricks Hub')
  File "C:\Users\johns\pybricksdev-demo\.venv\lib\site-packages\pybricksdev\ble\__init__.py", line 59, in find_device
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```

dlech, 2021-07-02:
Try this:

```python
    device = await find_device()
    await hub.connect(device)
```

If that doesn't work, enable Bleak logging: https://bleak.readthedocs.io/en/latest/troubleshooting.html

johnscary-ev3, 2021-07-02:
Ok that worked ;0)
Dropping the name string from the find_device() call fixed it.
Also the "program_running" boolean parameter you mentioned is working now, apparently replacing hub.state parameter.
The hub.write() method is still working, so I am back in business sending RC commands!
And there is a really neat status bar that shows up now while the program is downloading. 
Thanks for adding that and all the quick help on this.

closed 2021-07-02 by laurensvalk

laurensvalk, 2021-07-02:
> The hub.write() method is still working, so I am back in business sending RC commands!

We added it back just yesterday after you asked, so it sounds like you’ve found the latest version indeed 😄

johnscary-ev3, 2021-07-02:
Good timing for me ;0)
Thanks for putting write() back in and all the help.