pybricks/support #378: [Bug] Undefined Pybricks Code state after hub.system.reset(1)

laurensvalk, 2021-07-01:
**Describe the bug**
The run button can get stuck,  after `hub.system.reset(1)`, and stay that way even if you reconnect.

This test was done on the Technic Hub

**To reproduce**

[image]

First run the script, then click play again. 

```python
from pybricks.hubs import ThisHub
from pybricks.tools import wait

hub = ThisHub()

print(hub.system.reset_reason())

hub.system.reset(1)
```
And this isn't too bad. 

But it remains stuck after you reconnect, so a hard refresh is required.

[image]

**Expected behavior**
Disconnect as during a regular power-off.

laurensvalk, 2021-07-01:
Actually, the real cause is one step back. Even though the Bluetooth button correctly shows the disconnected state, the **run** button becomes enabled again.

[image]

So if this button were not enabled, the problem documented in the first post couldn't even occur.

dlech, 2021-07-01:
Fix is committed.

closed 2021-07-01 by dlech