pybricks/support #380: [Bug] Cannot find hub until removed from Bluetooth devices

laurensvalk, 2021-07-01:
**Describe the bug**

This never happened until some time ago, but now I've had it happen several times:

[image]

Version 91.0.4472.101 (Official Build) Built on Ubuntu , running on Ubuntu 18.04 (64-bit)
Pybricks Code (Beta) v1.1.0-beta.1

Just before that happens, I see this sometimes:

[image]

The problem goes away if I remove it from Bluetooth Devices:

[image]

Next time it happens, I can try to capture the logs.

dlech, 2021-07-01:
This is potentially due to the same cause as #374. BlueZ removes the Device Information service after connection to probe it and then adds it back. Chromium handles this, but there seems to be a race condition where things don't always work out. The recent addition of the PnP ID characteristic to the Device Information service seems to have made this worse.

dlech, 2021-07-06:
Since we added a new characteristic, everyone that uses Linux/BlueZ will probably have to remove the existing device from BlueZ when upgrading from v3.0.0 to v3.1.0. This is an unfortunate shortcoming in BlueZ.

dlech, 2021-08-31:
This happened to me due to setting

```
[GATT]
Cache = yes
```

in `/etc/bluetooth/main.conf`. 

In this case, it was due to a BlueZ bug which will be fixed in BlueZ 5.62. However, in my case, I had to remove the device from BlueZ after each time I connected it, so potentially not the same issue. But could still be due to the same underlying BlueZ bug which was related to the "UUIDs" property not being populated when scanning for devices. This can be confirmed by logging D-Bus traffic while reproducing the issue.

dlech, 2021-11-17:
I think we can close this one. It is all due to BlueZ bugs that are well understood by now and the workaround is documented at https://github.com/pybricks/support/discussions/270.

closed 2021-11-17 by dlech