pybricks/support #381: [Bug] Prime Hub stays on after software power-off if USB connected.

laurensvalk, 2021-07-01:
**Describe the bug**
When triggering a soft power-off while USB is connected, it does not shut down in the same way as holding the button.

The animation freezes and Pybricks Code stays connected (but fails to run anything).

**To reproduce**
```python
from pybricks.hubs import ThisHub
from pybricks.tools import wait

hub = ThisHub()

print(hub.system.reset_reason())
wait(100)
hub.system.reset(0) # Already applied support#379, so here doing power-off

# The run animation stays on and freezes
```

**Expected behavior**
Clean shutdown, same as holding the button.

**Screenshots**

dlech, 2021-07-01:
`0` is supposed to reset the hub, not power off. See #382 for power off.

laurensvalk, 2021-07-01:
See #379 as indicated in comment.

dlech, 2021-07-01:
We need to add a new method for proper shutdown that calls `pbsys_status_set(PBIO_PYBRICKS_STATUS_SHUTDOWN);`

laurensvalk, 2021-07-05:
This can be closed because `reset` will not be used. See #379.

See https://github.com/pybricks/pybricks-micropython/issues/58 for graceful shutdown.

closed 2021-07-05 by laurensvalk