pybricks/support #382: [Bug] Prime Hub does not turn back on after soft reset if USB disconnected.

laurensvalk, 2021-07-01:
**Describe the bug**
Soft reset does not work. It just turns off.

**To reproduce**
```python
from pybricks.hubs import ThisHub
from pybricks.tools import wait

hub = ThisHub()

print(hub.system.reset_reason())
wait(100)
hub.system.reset(1) # Already applied support#379, so here doing reset

# If USB is not connected, it does not turn back on.
```

**Expected behavior**
Turn back on, just like all the other hubs do.

dlech, 2021-07-01:
`1` means old-style instant power off without proper shutdown, not soft reset. We need to add a new method for proper shutdown that calls `pbsys_status_set(PBIO_PYBRICKS_STATUS_SHUTDOWN);`

laurensvalk, 2021-07-01:
See #379 as indicated in comment.

dlech, 2021-07-01:
Sorry, GitHub notifications lists the most recent issues first, the link wasn't clickable and I'm lazy.

We could try moving [this](https://github.com/pybricks/pybricks-micropython/blob/00f1fbc598a8306c0319066a1672263e08a10934/lib/pbio/platform/prime_hub/platform.c#L752-L758) and [this](https://github.com/pybricks/pybricks-micropython/blob/00f1fbc598a8306c0319066a1672263e08a10934/lib/pbio/platform/prime_hub/platform.c#L742) to the very beginning of `SystemInit`. If that doesn't work, then there is probably nothing we can do about it.

laurensvalk, 2021-07-02:
Soft reset will not be supported, so we can close this.

closed 2021-07-02 by laurensvalk