pybricks/support #383: [Doc] Document features not supported on Move Hub

laurensvalk, 2021-07-02:
Due to space constraints, the Move Hub will not support all MicroPython features.

Instead of adding special cases and notes throughout the documentation, we can make one dedicated page with features that the Move Hub does not have. To get started, let's collect them in this issue.

laurensvalk, 2021-07-02:
Unsupported MicroPython features
- REPL
- Floating point
- PYBRICKS_STM32_OPT_EXTRA_MOD (TODO: Make detailed list)

Unsupported Pybricks features
- iodevices (TODO: Drop support)
-

BertLindeman, 2021-07-02:
Do not know if ```help()``` belongs to the optional extra MicroPython modules above.
But the current firmware on the movehub does not support it.

dlech, 2021-07-02:
`help()` and other built-in functions fall under the category of `PYBRICKS_STM32_OPT_EXTRA_MOD `

laurensvalk, 2021-07-07:
We can also add some workarounds like https://github.com/pybricks/support/discussions/319

JorgePe, 2021-07-21:
> We can also add some workarounds like #319

Is there any workaround for `stdin` on MoveHub? I am using usys/uselect and keyboard.poll() fine with City Hub and Technic Hub and sending commands with `pybrickdev` hub.write()

laurensvalk, 2021-07-21:
> Is there any workaround for stdin on MoveHub?

Waiting for #218, probably, and hoping that we'll have enough space.

JorgePe, 2021-07-21:
Thanks.Will wait then.
But probably will postpone buying more Move Hubs in case "enough space" never happens.

Funny, I didn't like the Move Hub when I got my first... "uh, not another MicroScout".  But for the price, with 2 motors and a simple IMU sensor, it's quite a good deal when replacing batteries is not an issue.

dlech, 2023-03-23:
Support of individual APIs has been documented now for a while. Is there anything more to do here?