pybricks/support #396: [Bug] Keyboard regional settings wrong when using REPL with Chrome IDE

JorgePe, 2021-07-21:
I am using Chrome on Ubuntu. When using the REPL, some keys are wrong - like `(` and `)` that on a portuguese keyboard are located over `8`and `9`but now I need to use `9` and `0` instead.

Steps to reproduce the behavior:
1. Start Chrome and code.pybricks.com
2. Connect to a hub (I am using City Hub)
3. Click REPL
4. Write `help()`
5. You get `help*(`

[image]

 Expected behavior:
You should get `help()`

Using Chrome on Ubuntu `Version 92.0.4515.107 (Official Build) (64-bit)`
But on my VM, with a slightly older version of Chrome it works fine (91.0.4472.114)

writing code works fine, it only happens on REPL.
Also happens with older firmware releases, not just `v3.1.0a3`

dlech, 2021-07-21:
Apparently xcode.js hard-codes some stuff it shouldn't. I think this is basically the same as https://github.com/xtermjs/xterm.js/issues/2479.

dlech, 2021-07-21:
In particular: https://github.com/xtermjs/xterm.js/blob/5a4b305a68e1e49827f54c88ac6bd24a96cfcc83/src/common/input/Keyboard.ts#L21-L22

JorgePe, 2021-07-21:
so things like `(` and `)`are hardcoded for american-like keyboard mappings :(

dlech, 2021-07-21:
Yes, I was quite surprised to find that. Has this always been problem in Pybricks Code for you and you are just now getting around to reporting it (or didn't use the REPL before)?

dlech, 2021-07-21:
Digging a little more, it looks like that mapping is only used when the <kbd>Alt</kbd> key is pressed. So maybe this is a Chromium bug?

JorgePe, 2021-07-21:
> Yes, I was quite surprised to find that - especially since it is the same terminal used in VS Code. Has this always been problem in Pybricks Code for you and you are just now getting around to reporting it (or didn't use the REPL before)?

I did use the REPL before. Not much, mostly with Robot Inventor hub but did use it a few times this last year with Technic Hub to get parameters from motors - would have noticed it when printings things like `motor.control.limits()` to compare my motors and changing things like `motor.control.pid()` to get faster fingerings with the bagpiper.

VS Code is a different thing... I am still using plain ssh when coding for EV3, my brain cannot cope with so many things changing at the same time so I tend to keep doing somethings 'the good old even if inefficient way' 

I did not understand that part of the "only used when the Alt key is pressed"... to get `(` I only press `Shift` + `8`, no 'Alt' or 'Alt Gr'.

dlech, 2021-07-21:
> I did not understand that part of the "only used when the Alt key is pressed"... to get `(` I only press `Shift` + `8`, no 'Alt' or 'Alt Gr'.

Right. The hard-coded values I linked to are ignored in that case, so that is not the source of the bug.

dlech, 2022-06-03:
Is this still a problem? The upstream bug has been marked as fixed.