pybricks/support #401: [Feature] Implement pybricks.tools.Datalog for data logging with Pybricks Code

laurensvalk, 2021-07-23:
**Is your feature request related to a problem? Please describe.**
When used via `pybricksdev`, we have a hidden feature to allow a form of data logging.

Basically, after you print a magic command, all future standard output is written to a file. This can be quite useful for logging data, despite not having a file system on these hubs.

This can be very useful to log data or save execution results for further analysis on a computer.

**Describe the solution you'd like**
Enable something similar in Pybricks Code. The file gets downloaded just like we currently do with `main.py`. Perhaps when the script completes.

And instead of a magic command or hacking the real file API, implement the equivalent of our existing [`pybricks.tools.Datalog`](https://pybricks.com/ev3-micropython/tools.html#pybricks.tools.DataLog) class. This can work with Pybricks Code to appropriately create, write, and close multiple files.

Since the `.log()` method is essentially just a blocking `print` underneath, you can log vasts amounts of data without limitations of the hub's available RAM.

**Describe alternatives you've considered**
Use pybricksdev.

laurensvalk, 2022-08-23:
This was also requested below:

> in fact the DataLog class is for the EV3 hardware - we overlooked that in the heat of the moment.
>
> We would like to write log data from SPIKE - so will set a ticket accordingly. After all, we have no other way to do this.

_Originally posted by @legotronic in https://github.com/pybricks/support/issues/703#issuecomment-1217796851_

laurensvalk, 2022-11-12:
This was also requested below:

> I found DataLog module in EV3 but it is not working on spike prime. Also I cannot write or create csv file.
How do I log sensor data on Spike Prime?

_Originally posted by @mihye126 in https://github.com/pybricks/support/discussions/796_

LegoPink, 2024-04-21:
Datalogging (beyond the 1000 lines in the output window) would be very helpfull! 

Are you still considering this feature? 

Or, can you elaborate in the alternative you mentioned above, using 
pybricksdev?

laurensvalk, 2024-04-22:
If you run this script with `pybricksdev`

```python
print("_file_begin_", "./test.txt")

print("time, speed, angle")
print("0, 0, 0")
print("1, 10, 10")
print("2, 20, 30")

print("_file_end_")
```

It will produce a file `test.txt` on your computer as follows:

```
time, speed, angle
0, 0, 0
1, 10, 10
2, 20, 30
```

Anything between the begin and end "commands" will go into the file. It doesn't have to be columns with numbers.

> Are you still considering this feature?

In the long run, yes. It would be nice if could have a datalog class that will "print" data to a channel separate from the standard output pane. That way it won't get mixed with normal `print` statements, and it could write to different files at the same time.