pybricks/support #402: [Question] Does robotics.DriveBase work with PrimeHub/InventorHub?

AntoniLuongPham, 2021-07-23:
Dear @dlech and @laurensvalk,

I am trying to use the DriveBase code with the PrimeHub but it doesn't work.

Could you let me know if it works with only EV3 for now?

Thank you very much.

Antoni

dlech, 2021-07-23:
It should work. What is the problem? Is there an error message?

laurensvalk, 2021-07-26:
We could probably add an example that sets the direction correctly.

TheVinhLuong102, 2021-08-09:
David, Laurens,

I've looked further at the error that @AntoniLuongPham encountered the other day, and it actually didn't have to do with the DriveBase. Rather, he was using `set`s and got a `NameError: name 'set' isn't defined` on the PrimeHub.

I've asked him to change the title of this issue accordingly.

Thank you.

Vinh

laurensvalk, 2021-08-09:
Since we are using dual boot on these hubs, space is limited. It's not quite as bad as the restrictions on Move Hub, but we still have to carefully consider which features we enable.

See also https://github.com/pybricks/pybricks-micropython/pull/60, where we decided not to enable set, at least not for now.

TheVinhLuong102, 2021-08-09:
Thanks Laurens for the clarification -- I understand the rationale.

We were trying to use `set` because we were initially unsure about the order of the returned collection of `buttons.pressed()` whenn. After some more testing we confirmed for ourselves that `buttons.pressed()` returned a tuple with the "RIGHT" buttons preceding any "LEFT" buttons -- and hence able to get the job done with just tuple comparisons.

For now, I think these technical decisions should be documented so that the users of the Pybricks 3.x versions are aware.

Vinh

laurensvalk, 2021-08-09:
That's being done in #286 😉

TheVinhLuong102, 2021-08-09:
Brilliant!! -- looking forward to the formal release! :)

closed 2023-05-19 by dlech