pybricks/support #403: [Bug] The hub took too long to respond (Android)

laurensvalk, 2021-07-27:
**Describe the bug**
Firmware installation gets to 100%. Then after a few seconds this error pops up:

```
The hub took too long to respond
```

When you turn the hub on after this, the light blinks purple so there is no firmware.

This happens on Technic Hub and City Hub. (TODO: Test Move hub)

**To reproduce**
I used Pybricks Code v1.0.0 with Android 11 on a Moto G Pro. This is [Android One](https://en.wikipedia.org/wiki/Android_One) so fairly close to stock Android with little manufacturer customization.

**Related 1**

> I would rather start a new issue about Android since this one is about macOS.

_Originally posted by @dlech in https://github.com/pybricks/support/issues/329#issuecomment-857697355_

**Related 2**
https://github.com/pybricks/support/discussions/400 (Now deleted?)

dlech, 2021-07-27:
Can you log the Bluetooth packets when you repeat the problem?

PierreArch, 2021-08-07:
I am experiencing the same issue with my Samsung Tab S6 running Android 11.
I cannot complete the download to my Powerup hub. I works well with the LEGO Powerup app. I have made several attempts toggling between both environments with no avail.

dlech, 2021-08-07:
To solve this, we will need more information. See https://github.com/pybricks/support/discussions/270.

Chrome developer tools can be attached remotely. https://developer.chrome.com/docs/devtools/remote-debugging/ (this changed somewhat recently, so hopefully the docs have been updated)

Capturing Bluetooth packets on Android seems to change every few releases as well. On my Android 8 phone, I go to *System* > *Developer Options* and turn on *Enable Bluetooth HCI snoop log*. Then reboot for good measure. Then repeat the steps that caused the error. Then turn off the log. Then reboot. Then (on the same menu) *Take a bug report*. I don't remember which option from there, maybe *Full Report*. Then you can email the report to yourself, or something like that. Then buried in the report somewhere is the Bluetooth log.

laurensvalk, 2021-08-07:
> Can you log the Bluetooth packets when you repeat the problem?

I will. I might also try a few older builds of Pybricks Code to see if it got introduced there (vs. an Android update.)

PierreArch, 2021-08-11:
Hi,

 

Please find here attached, two different tentative I made at a few minutes intervals.

 

I hope this will help you resolve this issue. In the mean time, I can use my EV3.

dlech, 2021-08-11:
Github does not accept email attachments, so you will have to open the issue in a web browser and attach it that way.

PierreArch, 2021-08-12:
Here are the files. Tks

[btsnoop_hci.log.zip](https://github.com/pybricks/support/files/6975579/btsnoop_hci.log.zip)

dlech, 2021-08-12:
Thanks very much for this. From the logs, we can see the problem. The payloads are being truncated.
[image]

The size says how many bytes are being sent (including 4 bytes for the flash address), so the payload should be 36 - 4 = 32 bytes, but it is only 14 bytes.

I'm guessing that somewhere between Android 8 and Android 11, the default MTU negotiation behavior changed and it is using the smallest possible MTU (23 bytes) which limits the payload to 14 bytes (and size should be 20).

There is no way to get the MTU through existing web browser APIs. And we are just making the assumption that all platforms negotiate something higher, assuming the Powered Up hub supports it (the BOOST Move hub has max MTU of 23 anyway, so it should theoretically still work on Android 11 already).

Or maybe I have only flashed BOOST Move hub with Android and the comments about 8 vs. 11 are invalid? (The rest of the comments still apply though.)

So to fix, our options are:
1. Always assume the minimum MTU. This will make flashing firmware slower, but should always work.
2. Add a platform detection for Android. This will fix the immediate issue, but the problem could happen on other platforms in the future as well. And platform detection in web browsers is notoriously fragile as well.
3. Fix the Android MTU bug in Chromium itself. https://crbug.com/1164621
4. Implement getting the negotiated MTU in Chromium itself. https://github.com/WebBluetoothCG/web-bluetooth/issues/383

dlech, 2021-08-12:
It also seems odd that we aren't getting an error from a bad checksum since the hub is not receiving all of the data. So there could be a bug there too.

dlech, 2021-08-12:
I think 2. in the short term and 3. and/or 4. in the long term could be feasible.

laurensvalk, 2021-08-12:
Regarding your question about move hub working in theory, this is also true in practice (see first post in this thread).

And option 2 as a first step seems like a good plan 👍 

And thanks @PierreArch !

dlech, 2021-08-13:
> Or maybe I have only flashed BOOST Move hub with Android and the comments about 8 vs. 11 are invalid?

I seem to have some faulty memory modules between my ears. I was able to reproduce on Android 8 and confirm the fix works.

PierreArch, 2021-08-13:
I am glad I could help by sharing the files, how can I use this fix now? I am new to GitHub and I am not sure what to do next. I tried the app using the same shortcut and it didn't work ! 
Please advise,

dlech, 2021-08-13:
As noted in the [changelog](https://github.com/pybricks/pybricks-code/blob/master/CHANGELOG.md), the fix is still unreleased. But I'm actually planning on doing a release today, so you won't have to wait long.

dlech, 2021-08-13:
Fix is released in Pybricks Beta v1.1.0-beta.4.

laurensvalk, 2021-08-16:
> Fix is released in Pybricks Beta v1.1.0-beta.4.

Confirmed working :rocket:, so we can probably close this.

closed 2021-08-16 by laurensvalk