pybricks/support #404: [Feature] Add "address" and "light" to class "remote"

falk12, 2021-07-27:
Hi all,

the features "address" and "light" are discussed/planned , but yet not implemented in class "remote"

Use of "address"
- connect desired remote to the hub
- reconnect same remote to hub as before
- espacially on exhibitions where you have many active devices  

Use of "light"
- show different modes of the program/remote
- i.E. switch from Port A/B to B/C with a technic-hub via "center" button
- i.E. alarm battery level of hub connected to
- i.E. status of devices etc.

Perhaps discuss a "disconnect" for the remote?

dlech, 2021-07-27:
> Perhaps discuss a "disconnect" for the remote?

What is the use case for this?

falk12, 2021-07-27:
> What is the use case for this?

- drive a train, stop it, disconnect, start another train with same remote
- having a GBC with many motors running with a certain speed, but connect and disconnect one remote to the different hubs to adjust speed (by pressing hub button and remote center simultaniosly)

laurensvalk, 2021-07-28:
> Perhaps discuss a "disconnect" for the remote?

:+1: 

Another case was even used in our own [video](https://youtu.be/kN47P6mksj4?t=74). But since disconnect() was not available I  cheated and just turned the whole hub off instead :smile:

laurensvalk, 2021-08-05:
Setting the light color seems relatively slow. Would it be faster if we didn't use RGB mode, or is two extra bytes negligible given the overhead data?

laurensvalk, 2021-08-05:
As a minor note, `remote.light.on(Color.YELLOW)` is pretty much green. More so than, say, Technic Hub.

laurensvalk, 2021-08-05:
The remote seems to remember the light used last. Maybe we can add call `self.light.on(Color.WHITE)` Remote.__init__() to get a clean state, which is consistent with re-connecting when re-starting the script.

dlech, 2021-08-31:
This is available at https://beta.pybricks.com now.

closed 2021-08-31 by dlech

falk12, 2021-08-31:
@dlech 

yes indeed, **thanks a lot!**
But I expected and hoped that you could specify a MAC-Address when connecting to be sure to have teh appropriate one ?
No?
So I would have to rename remote to MAC

dlech, 2021-08-31:
It is possible to give each remote a unique and meaningful name. We hoped this would be nicer than using the Blueooth address.

And if you can be sure that only one remote is advertising at a time, then you don't need the name at all.