pybricks/support #405: [Doc] Online Doc does not reflect some functions

falk12, 2021-07-27:
Could not find hub.system nor hub.button online, but in 
docs-pybricks-com-en-latest.pdf

dlech, 2021-07-27:
When you say "online" do you mean at https://code.pybricks.com or https://docs.pybricks.com?

falk12, 2021-07-27:
> When you say "online" do you mean at https://code.pybricks.com or https://docs.pybricks.com?

I meant https://code.pybricks.com

dlech, 2021-07-27:
Got it. (Those docs are actually downloaded to your computer, so technically offline.) They only get updated when we release a new version of Pybricks Code.

falk12, 2021-07-27:
I see, that explains, good to know.
THX

laurensvalk, 2021-07-28:
Thanks for opening this issue. We can make this less confusing by making the `stable` docs the default, instead of the latest `master`.

The current [stable](https://docs.pybricks.com/en/stable/) version seems to be pegged at (the now outdated) 2.0, so we'll need to fix that first.

laurensvalk, 2021-07-28:
It is now possible to pick a version at the bottom left:

[image]

The default version will be `stable` (equal to the highest number) instead of `latest`.

Starting with the next release (`v3.1`), the "online" and "offline" docs will become consistent again.

closed 2021-07-28 by laurensvalk