pybricks/support #406: [V3.2] SPIKE Prime Hub and MINDSTORMS Inventor Hub (Ideas wanted!)

laurensvalk, 2021-07-28:
**Introduction**

An early version of Pybricks for SPIKE Prime and MINDSTORMS Robot Inventor has been working for a *long* time. But so far it was only aimed at adventurous users who didn't mind trying out nightly builds with cumbersome installation [tools](#167).

But now, we are beginning to finalize a stable release for everyone. :rocket: 

[image]

**SPIKE and MINDSTORMS support will come in V3.2!**

Here's what we'll focus on at first. Some of this is even already done:

- Easy and quick firmware installation.
   - Using USB via our online app.
   - Installed alongside the original firmware. You can pick which one to start using the buttons.
- Download programs via BLE and save them on the hub.
- Fast program start, fast code execution, and increased memory availability.
- Charge the hub via USB.
- Support for all motors and sensors:
  - All SPIKE/MINDSTORMS motors and sensors.
  - All Powered Up motors and sensors.
  - Powered Up remote.
- Support for reading all buttons, including the center button.
- Support for the light matrix display, including animations.
- Basic support for the built-in IMU sensor.
- Complete documentation of all Pybricks modules and all MicroPython modules.

Hub-to-hub communication will be reserved for V3.3.

**We want your input!**

As always, a lot more is possible. But what features do you really want to see?

Let us know in a comment below. Or tell us whether we should just stick to this list for now.

**Upvote!**

To let us know that you're here, please :+1: this post. This really helps us prioritize user requests. This was a big hit in #186 --- let's see if we can set a new record.

TechnicBRICKs, 2021-07-28:
This list of intended features is great!
I'm most interested in the Hub2hub communication intended for V3.3.

I'm currently focusing at other home needed activities for a few months, so hopefully when I find the time to work on my robots again this will be much closer. :)

JorgePe, 2021-07-28:
'Download programs via BLE and save them on the hub' and 'Charge the hub via USB' are very welcome.
A stable official version for SPIKE/RI will be great, will bring more attention to the whole ecosystem.

Don't mind to wait for Hub2Hub communication... there are already enough features on current release to keep me occupied :D

TheVinhLuong102, 2021-08-07:
I second @JorgePe re: "Download programs via BLE and save them on the hub", and would like to request something more ambitious: the ability to access / manipulate the Hub's filesystem (user space only, not root/system space).

The use case I have in mind is to enable complex projects with multiple .py files with mutual imports (e.g. `main.py` importing some common helpers from `util.py`), and external data files (e.g. JSON files).

The standard to reference against is what is currently possible with EV3 through VSCode, i.e., downloading a project of multiple files to the EV3 Brick's filesystem, and maintaining multiple projects on the EV3 Brick.

Vinh

TheVinhLuong102, 2021-08-07:
Talking of the VSCode experience, I notice this active project by Peter Staev: https://github.com/PeterStaev/lego-spikeprime-mindstorms-vscode <-- referenced here for what it's worth.

HvMidden, 2021-08-13:
The list of suggestions is great, I hope it will come soon

Vinz1911, 2021-08-18:
Saving programs on the hub and charging the hub via USB are also for me the essential features here. I flashed Pybricks directly to get completely rid of the original firmware (I don't need dual boot). Personally I'm not a big fan of the web IDE but I'm a big fan of pybricksdev. So it would be very nice to access the hub via REPL Mode and stop programs via the terminal.

lobodpav, 2021-12-31:
For a better development experience and code maintenance, it would be great to allow splitting the code into multiple modules/files to avoid a single bulky python source file.

A typical scenario is a re-use of custom-made tools/libraries for multiple projects.

jriise, 2022-01-13:
It would be cool if you could implement the hub style commands for motors (hub.port.A.motor)

laurensvalk, 2022-01-14:
We've just [released](https://github.com/pybricks/support/issues/591) a beta version that has pretty much all of these features in one form or another.

So we can close this issue.

Feel free to keep your ideas coming --- just open a dedicated issue or discussion for each one.

closed 2022-01-14 by laurensvalk

lobodpav, 2022-01-14:
> We've just [released](https://github.com/pybricks/support/issues/591) a beta version that has pretty much all of these features in one form or another.
> 
> So we can close this issue.
> 
> Feel free to keep your ideas coming --- just open a dedicated issue or discussion for each one.

Sounds great! Does it also include the `multiple modules/files` suport?

laurensvalk, 2022-01-14:
No, please see https://github.com/pybricks/support/issues/591, where this is mentioned.