pybricks/support #409: [Bug] PrimeHub/InventorHub: hub.display.image(...) argument mismatch b/w firmware & documented API

TheVinhLuong102, 2021-07-30:
Hi @laurensvalk, @dlech,

For the PrimeHub/InventorHub, the method `.display.image(...)`'s argument name in the documented API is `matrix=...`, but the underlying firmware accepts `image=...` instead.

Vinh

dlech, 2021-07-30:
Off topic: Can we save the @ mentions for when you really need to get our attention (like if it has been 3 or 4 weeks and we haven't responded)? Each time we are @ mentioned, it creates an extra notification.

TheVinhLuong102, 2021-07-30:
Ok noted David! 👍

dlech, 2021-07-30:
Thanks :smile:

laurensvalk, 2021-08-06:
Since the Prime Hub isn't part of a release yet, we still have some flexibility to reconsider the API.

We've currently grouped the "images" under `pybricks.parameters.Icon`, so it's currently like this:

```python
hub.display.image(Icon.HAPPY)
```

I called them icons because at 5x5 pixels it's a bit of a stretch to call them images.

So another option that could make sense is

```python
hub.display.icon(Icon.HAPPY)
```

TheVinhLuong102, 2021-08-06:
Laurens,

Yes I think "Icon" / "icon" sounds great. It's better to settle on one way to call these -- especially for young users -- instead of having "icon" + "image" + "matrix" :)

Vinh

laurensvalk, 2022-11-22:
> So another option that could make sense is

   ```python
   hub.display.icon(Icon.HAPPY)
   ```

This is done. Thanks @TheVinhLuong102!