pybricks/support #410: Installing and using Pybricks under Linux

jeagh, 2021-08-01:
A connection dialog pops up when I attempt to install Pybricks, Robot Inventor version, but a message tells me that a compatible devise cannot be found. lsusb shows that a device named Lego Group has been found. I also tried Edge under Windows and the connection dialog does show that my Lego hub has been found.

I'd like to install Pybricks but ONLY if it's usable under Linux. So, if I do continue with the installation using the Windows browser will I then be able to use Pybricks under Linux? Are there any Linux users happily using Pybricks?

dlech, 2021-08-01:
Which website are you using? https://beta.pybricks.com works on Linux. There is nothing to install.

jeagh, 2021-08-01:
Thank you dlech for your reply.

I had been communicating with someone via FaceBook, who had signed in as pybricks. I was directed to this link https://install.pybricks.com/. An explaination was given on how to to use the dual boot mode. Now I'm really confused now that you tell me that I don't have to install Pybricks onto the hub.

I went to the https://install.pybricks.com/ and it seems that I cannot proceed until I have installed Pybricks onto the hub.

dlech, 2021-08-01:
Sorry, I meant that you don't have to install Pybricks on your computer. You still have to flash the firmware to your hub. You probably need to install [these udev rules](https://github.com/pybricks/pybricksdev/blob/master/pybricksdev/resources/99-pybricksdev.rules) on Linux to allow permission to access your hub.

You may need to ensure that your hub has the latest firmware from LEGO first.  These are the known working official LEGO versions: 

https://github.com/pybricks/pybricks-micropython/blob/9443437f44cf912cd1507e4b20fa967fa7320a8b/bricks/primehub/install_pybricks.py#L32-L34

jeagh, 2021-08-01:
Sorry to be a bit slow, do I simply copy

  # MINDSTORMS Inventor hub SUBSYSTEM=="usb", ATTRS{idVendor}=="0694", ATTRS{idProduct}=="0010", MODE="0666"

# MINDSTORMS Inventor hub in firmware update mode (DFU mode)
SUBSYSTEM=="usb", ATTRS{idVendor}=="0694", ATTRS{idProduct}=="0011", MODE="0666"

to "/etc/udev/rules.d/99-pybricksdev.rules"? The use of the script is not clear to me.

I do have the latest Lego firmware.

dlech, 2021-08-01:
yes

dlech, 2021-08-01:
If the hub is plugged in while you do this, unplug it and plug it back in for the changes to take effect.

jeagh, 2021-08-01:
More problems dliech.

This is the message that I received using Firefox or Chromium. What's the recommended browser?

"This web browser does not support Web Bluetooth or it is not enabled."

dlech, 2021-08-01:
Chromium is required. You also need to enable experimental web platform features on Linux.

Paste this in the address bar of Chromium: `chrome://flags/#enable-experimental-web-platform-features`

jeagh, 2021-08-01:
Chromium experimental features are now enabled and I'm back to "no compatible device found" Is the following correct? I've tried with and without the "#" and I've rebooted.

more /etc/udev/rules.d/99-pybricksdev.rules
#MINDSTORMS Inventor hub SUBSYSTEM=="usb", ATTRS{idVendor}=="0694", ATTRS{idProduct}=="0010", MODE="0
666"

#MINDSTORMS Inventor hub in firmware update mode (DFU mode)
SUBSYSTEM=="usb", ATTRS{idVendor}=="0694", ATTRS{idProduct}=="0011", MODE="0
666"

dlech, 2021-08-01:
You are missing a newline in the file. I edited your comment above with the correct format.

jeagh, 2021-08-01:
Thank you for being so patient.

I cannot see any difference in my udev file. There's a newline between the two #MINDSTORM comments.

I think that I'll have to give this away, for the time being at least.

jeagh, 2021-08-01:
I cannot leave this alone. I see what you mean and I have corrected the newline error but I'm still getting a "no compatible device error" message. Is there a way to test the rules, probably not? 

lsusb shows this "Bus 001 Device 007: ID 0694:0010 Lego Group"

The hub is displaying the "play" symbol, is that correct?

laurensvalk, 2021-08-01:
Thanks for opening this issue here. Much easier for us to keep track than on Facebook.

> I'd like to install Pybricks but ONLY if it's usable under Linux. So, if I do continue with the installation using the Windows browser will I then be able to use Pybricks under Linux?

Yes. Everything besides the installation uses Bluetooth Low Energy. So if your computer has that, it should work.

> Are there any Linux users happily using Pybricks?

Yes, quite a few, myself included. 😄

jeagh, 2021-08-01:
I think I've reached the end of my ability, Laurensvalk, to proceed any further. lech has offered many suggestions but I still cannot get a connection to install the firmware. Can you see anything in my ramblings today in this thread that might offer a clue to get the installation under way?

laurensvalk, 2021-08-01:
I believe you mentioned above in that quote that your Windows machine did let you connect via USB on that installation page. But I might have misunderstood. 

If that's the case, you may be able to install there, and then come back to Linux to do the actual coding at beta.pybricks.com.

I appreciate you taking the time to help us identify issues like these. This will really help us making the final version a lot better.

And definitely feel free to leave it for now if you like. We'll respond here once we've had a chance to look at it in detail and let you know if there's something new to try.

jeagh, 2021-08-01:
Thank you for your reply laurensvalk.

My laptop is dual boot so the only difference that I can see is the Linux udev rules. The connection dialog does display under windows but I'm afraid that if I install the firmware on my hub under Windows that I will still not be able to us Pybricks under Linux.

laurensvalk, 2021-08-01:
Those two things (installation and coding) are fully independent, so it could be worth a try, but no problem if you prefer to wait for a Linux fix.

closed 2021-08-15 by jeagh