pybricks/support #411: [Bug] Running scripts located in "pybricks-micropython/test" folder with pybricksdev results in SyntaxError

NStrijbosch, 2021-08-02:
**Describe the bug**
Running scripts located in "pybricks-micropython/test" with pybricksdev results in SyntaxError

**To reproduce**
Steps to reproduce the behavior:
1. Open terminal in pybricks-micropython folder
2. type: 'pybricksdev run ble /tests/pup/hub/status_light.py'
3. press enter

**Expected behavior**
I expected the program to be executed on the hub without an issue. When creating a new file in a different folder and copying the script's content, there is no error when using the command 'pybricksdev run ble ... ' and the script is executed without an issue.

**Screenshots**
[image]

dlech, 2021-08-02:
>     2\. type: 'pybricksdev run ble /tests/pup/hub/status_light.py'

If there is a leading `/` before `tests`, then it is an absolute path instead of relative path. So I would expect some sort of "file not found" error.

Since the command will also compile text to code, it is probably treating the text as the contents of a file instead of a file name.

Dropping the leading `/` should fix the immediate problem. But I think adding a command line argument to differentiate between filenames and text would be a good idea.

dlech, 2021-08-02:
Another option could be to make it more like traditional unix command line tools and allow piping instead (where a file name of `-` is shorthand for stdin).

```shell
$ echo 'print("hello")' | pybricksdev run ble -

$ pybricksdev run ble - <(echo 'print("hello")')

$ cat <<EOF > pybricksdev run ble -
print("hello")
print("goodbye")
EOF
```

dlech, 2021-08-02:
Then you can do fun stuff like run the first 20 lines of a program :grin: 

```shell
$ head -n 20 program.py | pybricksdev run ble -
```

laurensvalk, 2021-08-02:
> But I think adding a command line argument to differentiate between filenames and text would be a good idea.

Or we just allow only files, and I guess that includes the "fun stuff" above.

dlech, 2021-08-02:
> Or we just allow only files,

If you are OK with that, I'm all for it.

laurensvalk, 2021-08-02:
Yes, let's drop it. It seemed like fun when I made it, but I haven't used it since. It's also not terribly practical with all the quotes on the command line.

NStrijbosch, 2021-08-02:
> Dropping the leading `/` should fix the immediate problem.

Yes it did! Thanks. 

My mistake.. but indeed a 'file not found' error would have been more helpful.

closed 2021-08-04 by dlech