pybricks/support #412: [Feature] Show firmware version in Pybricks Code about menu

laurensvalk, 2021-08-02:
**Is your feature request related to a problem? Please describe.**
Pybricks Code comes with a specific pybricks-micropython firmware release preloaded, but it's hard to know which one since the versioning schemes don't match.

In documentation, as well as roadmaps like #286 we usually express versions in firmware terms (V3.1, V3.2), but this not translate to something that the user can easily see.

**Describe the solution you'd like**

Also show the bundled pybricks-micropython firmware version on here somewhere. Perhaps show both in the title.

If the firmware version scheme is the one we communicate everywhere, it would make sense to show that first, e.g.:

**Pybricks 3.1.0a3 (Pybricks Code v1.1.0-beta.3)**

[image]

Perhaps we can improve the changelog button as well by having both `Firmware Changelog` and `Editor Changelog`. It defaults to the editor now which, much like the versioning above, probably isn't the changelog that most users would be most interested in.

dlech, 2021-08-13:
> Perhaps we can improve the changelog button as well by having both `Firmware Changelog` and `Editor Changelog`.

Having too many buttons gets ugly because they start wrapping around. So we should use something other than a button or wait for https://github.com/pybricks/pybricks-code/issues/468. The current changelog links to the firmware changelog anyway, so it only potentially saves one click.

Or we could just copy the firmware changelog to the pybricks code changelog when we update it.

dlech, 2021-08-13:
Fix for the primary issue is released in Pybricks Beta v1.1.0-beta.4.

We can leave this open or start a new issue to discuss the changelog.

dlech, 2021-08-30:
I opted to copy the firmware and docs changelogs to the Pybricks code changelog since it lets users see everything in one place and involves the least amount of clicking.

closed 2021-08-30 by dlech