pybricks/support #414: [Doc] Collapse compatibility tables

laurensvalk, 2021-08-03:
**Describe the issue**
We're currently working on compatibility tables to show on which hub a particular feature works:

[image]

They're fine, but they can be distracting for most users if we show them everywhere. Showing them only in some places sort of works, but it still leaves you wondering what is the default. And sometimes this leads to awkward attempts to categorize things, making things harder to find.

If they were completely unobtrusive, we could use them more liberally including for particular quirks within functions, like the missing `file` argument on `print`.

**Suggested improvement 1**
Make them folded by default, with an easy way to unhide them. We had this before using sphinx extension. We dropped it for content because it did not work great for permalinks. But it could be fine for this kind of information.

**Suggested improvement 2**
As above, but show an unfold symbol next to the heading, much like the permalink icon. This makes it take even less space for most users.

laurensvalk, 2021-08-03:
> As above, but show an unfold symbol next to the heading, much like the permalink icon. This makes it take even less space for most users.

This seems like the nicest, but might involve a bit too much hacking of Sphinx at this point. Something like [this](https://stackoverflow.com/a/29578663) might work with a bit less effort.

laurensvalk, 2021-08-03:
:sunglasses: 

https://user-images.githubusercontent.com/12326241/128004961-0c2f1ad4-ba61-4f21-9b67-455100c41724.mp4

XilefTech, 2021-08-04:
how about adding the EV3 as well? I mean pybricks does support EV3 so wouldn't it be fair to add EV3 to the compatibility list as well?

laurensvalk, 2021-08-04:
For the foreseeable future, Pybricks for EV3 will likely stay at version 2.0, which has its own [dedicated version of the documentation](https://pybricks.com/ev3-micropython/).

EV3 support is already quite complete, so that's why ​we're focusing on the other hubs.

And since EV3 is quite different from the other hubs, having separate documentation is much easier to read and maintain.

laurensvalk, 2021-08-06:
Done and merged to master, so closing.

closed 2021-08-06 by laurensvalk