pybricks/support #417: [Bug] Connecting to remote suppresses standard output

laurensvalk, 2021-08-04:
**Describe the bug**
Some output gets suppressed after connecting to the remote and reading the name.

Changing the name also appears to fail.

This happens consistently, so may provide some debugging starting points,

**To reproduce**

```python
from pybricks.pupdevices import Remote
from pybricks import version

print(version)

# Connect to any remote.
print("Connecting...")
my_remote = Remote()

print("Connected to:") 

# Print the current name of the remote.
print(my_remote.name())

# Choose a new name.
my_remote.name('truck2')

print("Done!")
```

Gives

```
('technichub', '3.1.0a3', 'v3.1.0a3-37-g3c277462 on 2021-08-04')
Connecting...
et
Done!
```

**Expected behavior**
```
('technichub', '3.1.0a3', 'v3.1.0a3-37-g3c277462 on 2021-08-04')
Connecting...
Connected to:    <----- This is missing altogether
handset:  <----- Probably. Seems like we see last two characters. And it's the same despite changing it.
Done!
```

dlech, 2021-08-04:
> And it's the same despite changing it.

This is probably a separate issue. I think this happens because we disconnect the handset at the end of the program. When we send a message, it is using write without response. We only block until this message has been sent to the local Bluetooth chip. But there are already other messages queued there from printing to stdout. The disconnect command probably doesn't wait for these queued messages to be sent and drops them. If you add a delay to the end of the program, the name will change as expected. Printing to stdout still works after the program has ended since that is a different connection that is not disconnected.

So I think the solution for that one is to figure out if there is any kind of feedback from the Bluetooth chips to see if all pending messages have been sent before disconnecting after the user program has ended.

Another possibility could be to use write with response instead of write without response. This would make communication slower since we always have to wait for the response. And the user would have to handle the error if they don't want the program to crash when a message failed.

dlech, 2021-08-17:
This is fixed on City and Technic hubs now (fix for #432 fixes this). Similar changes still need to be made to the Move hub.

dlech, 2021-08-18:
Fix for Move hub has been committed now as well.

dlech, 2021-08-18:
It looks like this is not entirely fixed. The original test case works not, but when I run the following program on the Move hub, the output ends at `Connected to: truck2` with no newline. When the program ends, the IDE doesn't update (stop button is still active). We known the program has ended because pressing the button on the hub starts the in-flash-memory program, but the IDE still doesn't respond (meaning status messages are not being sent/received). Disconnecting the hub from the IDE fixes everything.

```python
from pybricks.hubs import ThisHub
from pybricks.pupdevices import Remote
from pybricks import version
from pybricks.parameters import Color
from pybricks.tools import StopWatch

print(version)

hub = ThisHub()

print("Connecting...")
remote = Remote()

print("Connected to:", remote.name()) 

timer = StopWatch()

for h in range(360):
    c = Color.BLUE << h
    remote.light.on(c)
    hub.light.on(c)

print(timer.time())

print("Done!")
```

BertLindeman, 2022-10-18:
Copied from #736 

Print output not OK on the MoveHub. 

Even on a simple print does not get correct output.
Example:

```python
print("          1         2         3         4         5         6         7         8")
print("0123456789 123456789 123456789 123456789 123456789 123456789 123456789 123456789 ")
```

produces:
```
          1         2         3         4         5         6         7         8
0123456789 123456789 123456789 123456789 123456789 123456789
789
```

With a ```wait(20)``` after each print, the output is correct:
```
          1         2         3         4         5         6         7         8
0123456789 123456789 123456789 123456789 123456789 123456789 123456789 123456789
```

dlech, 2023-03-23:
After some reworking of the BlueNRG driver, the reproducible case from https://github.com/pybricks/support/issues/417#issuecomment-901386585 is fixed. There were probably multiple contributing factors to the various symptoms, but the commit https://github.com/pybricks/pybricks-micropython/commit/04d548aca9c3b03c74430bb5324cfe57856ba4ca is the first commit where the symptom "When the program ends, the IDE doesn't update (stop button is still active)." is fixed.

closed 2023-03-23 by dlech