pybricks/support #418: [Feature] multiple tabs on PyBricks Code IDE

afarago, 2021-08-05:
**Is your feature request related to a problem? Please describe.**
I have had many example codes prepared for myself and for kids to demo. I did use copy paste from Notepad++ for sake of simplicity.

**Describe the solution you'd like**
I would like to see many tabs opened and saved by PyBricks Code, similarly to Notepad++.
Additional geek goodies would be github integration with commit/push.
Additional geek goodies would be more kids collaborating on one file e.g. via github.

**Describe alternatives you've considered**
Notepad++ copy-paste or file open/save.

**Additional context**
n/a

dlech, 2021-08-09:
Related: #84 and #173

afarago, 2021-09-10:
Discussion on the #465 as well with @laurensvalk it seems that basically there would be one editor instance anyhow open, while in a background store (localstore?) app would store all contents of other source files

list of files -> vertical list representation (at its simplest form) or tabs -> horizontal list representation (at its simplest form)

(I will use "file" for a string of program contents that can be opened in an editor window)

as a next step (assuming) the concept of multiple source is accepted I would need help / guidance with the following:

* where to store the file data - in a prototype I did implement I am using "localstore_tab1", "localstore_tab2", ... - is it acceptable to append a suffix to the current localstore key name?
* where to store the list of available files - is this to be stored in an additional localstore item?
* UX representation of files (tabs, list of files on the left, etc)
* on local Open or Save action how to pass the file name to and from sagas/Editor - is this Redux or something else

[image]
[Editor.zip](https://github.com/pybricks/support/files/7142471/Editor.zip)

dlech, 2021-09-10:
> it seems that basically there would be one editor instance anyhow open,

If we get hub-to-hub communication working, then I would expect people to have multiple Pybricks Code windows open - one for each hub. In this case, would both windows have the same tabs or independent tabs (e.g. if you create a new tab in one window, does the other window get it too?)? And what happens when the app is closed and opened again? It seems natural to restore the tabs that were open before, but if each window has independent tabs, which ones are restored?

laurensvalk, 2021-09-10:
> If we get hub-to-hub communication working, then I would expect people to have multiple Pybricks Code windows open - one for each hub. In this case, would both windows have the same tabs or [...]

I was chatting with @afarago earlier about this issue. He was also interested in [this](https://github.com/pybricks/support/issues/465#issuecomment-916739590) as a possible alternative to tabs.

Perhaps we can use this issue to discuss some of the technical questions about file storage raised above.

dlech, 2021-09-10:
>     * where to store the file data

We should probably make an abstraction for this so we can change it easily. For now, we can continue to use localStorage if that is easiest (indexDB could be an alternative as it might making querying existing files easier).

>     * is it acceptable to append a suffix to the current localstore key name?

I think a prefix might make more sense, then all of the files will be grouped together when looking at localStorage in the dev tools.

Going with the abstraction idea, we could treat the name of each Python program as a relative file path (this will allow for future multi-file projects). In the abstraction layer, we refer to files by the path. Then the localStorage implementation would add a prefix to create the localStorage key, e.g. "file:myfile.py", or something like that.

> where to store the list of available files - is this to be stored in an additional localstore item?

It looks like we can iterate local storage keys ([example](https://developer.mozilla.org/en-US/docs/Web/API/Storage/key#examples)) so I don't think we would need to store a separate list of files. 

> on local Open or Save action how to pass the file name to and from sagas/Editor - is this Redux or something else

Redux (as part of an Action object).

dlech, 2021-09-10:
Another thing to consider before implementing any UI is https://github.com/pybricks/pybricks-code/issues/468. If we are going to make the change to a different UI toolkit, we should do it now before adding new UI.

afarago, 2021-09-16:
I have created an initial concept on a feature branch on a fork for pybricks. (I a noob working with github - please just tell me if I am not on the right path).

At this point I have a draft for
* reading the list of proper localstore items
* creating a project tab on the left
* effectively switching and syncing with the localstore on editor data change (user types code in)
* loading-saving active tab

So my problem is that when OpenFileButton fires onFile --> routed to editor/sagas/open -- there is a reference to the monaco editor (I guess this was and architectural decision).
I am already passing the file name along with the EditorOpenAction

* Yet I would need some reference to the Editor instance to open a new "tab" (handle left menu)
* or even have a higher level restructuring creating a new component above the editor and incorporate the editor and the "project tab" to make a proper abstraction. -- yet my React.js knowledge is far below as of today.

Could you set me on the right path to make the next steps?

dlech, 2021-09-16:
It would be easiest to review and comment on the code if you make a pull request (there is even a feature to mark the pull request as a draft to indicate that it is a work in progress and not ready to merge yet).

dlech, 2022-06-03:
This feature has been released on <https://beta.pybricks.com>.

For the additional geek goodies, we would recommend creating a VS code extension instead of adding the to Pybricks Code.

closed 2022-06-03 by dlech