pybricks/support #419: [Bug]

SpinCraft95, 2021-08-06:
```
GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
```

dlech, 2021-08-06:
Can you please give more details? What Operating System system and version do you use? What web browser and version do you use? What are the exact steps to reproduce the problem?

SpinCraft95, 2021-08-06:
I use windows 10 pro 20h2 and chrome 92 adn when i whant to connect the hub i do this :
-I click the green button and wiat the purple light
-i click on install pybrick firmware and select my technic hub
-it turn on green blue red and this error appeared

dlech, 2021-08-06:
Sounds like a duplicate of #312. If you have any more insights, please reply there.

Also see https://github.com/pybricks/support/discussions/270 for some troubleshooting tips.

closed 2021-08-06 by dlech