pybricks/support #420: [Question] Flash PrimeHub with pybricksdev on Mac ?

Vinz1911, 2021-08-06:
Hi @dlech @laurensvalk 

I have a question about the flashing process for the PrimeHub. Personally I'm not a big fan of chrome or using a web ide. I write my code directly on my Mac. so i played a little bit with pybricksdev and it work like charm with the technic hub. I would like to know If it's possible to also work with the Primehub and pybricksdev. I installed dfu-util but I run in an issue (I used dfu restore + technic hub firmware <- is this the correct firmware?) : 

>dfu-util 0.10
>
>Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
>Copyright 2010-2020 Tormod Volden and Stefan Schmidt
>This program is Free Software and has ABSOLUTELY NO WARRANTY
>Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
>
>dfu-util: Warning: Invalid DFU suffix signature
>dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
>dfu-util: No DFU capable USB device available

Maybe you can light up me here :) Thanks and best regards.

dlech, 2021-08-06:
I've just pushed a potential fix. I haven't tested on Windows or Mac yet.

> I used dfu restore + technic hub firmware <- is this the correct firmware?

No, you need to use the `primehub` firmware with the SPIKE Prime hub.

The usual flow is:

```shell
# back up the official LEGO firmware
pybricksdev dfu backup my_backup.bin
# flash the Pybricks firmware
pybricksdev flash primehub-firmware.zip
```

It doesn't look like dual-boot is an option with pybricksev at this time.

Vinz1911, 2021-08-06:
Thank you for the fast reply :) 
I'm a bit confused, where can I find the latest build for the Primehub? 

I will try, thank you very much. That's not a problem, I don't need the Dual Boot Option.

dlech, 2021-08-06:
You can find the latest builds here: https://github.com/pybricks/pybricks-micropython/actions/workflows/build.yml?query=branch%3Amaster

Click on the most recent (top) build  and scroll down to "Artifacts".

Vinz1911, 2021-08-07:
Hi @dlech 

Your fix is working well :) It was possible to flash the firmware now without any problems. 

> dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0694:0011
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "LEGO LES HUB "
Downloading element to address = 0x08008000, size = 342977
Erase   	[=========================] 100%       342977 bytes
Erase    done.
Download	[=========================] 100%       342977 bytes
Download done.
File downloaded successfully

But it was not possible to start the Primehub just with the Pybricks firmware for the Primehub. Do we need the Dualboot to run the Pybricks Firmware ? 

But I think the Bug can be closed :)

dlech, 2021-08-07:
What command did you run?

Vinz1911, 2021-08-07:
i flashed it with: 

`pybricksdev dfu restore primehub-firmware.zip`

dlech, 2021-08-07:
That will just copy the .zip file directly, which doesn't work. Use `pybricksdev flash primehub-firmware.zip` instead.

dlech, 2021-08-07:
`pybricksdev dfu restore my_backup.bin` will restore the firmware backed up with `pybricksdev dfu backup my_backup.bin`.

Vinz1911, 2021-08-07:
Awesome! Everything works now as expected :) 

Thank you!!

closed 2021-08-07 by Vinz1911