pybricks/support #425: Powered Up Handset status light inital color

dlech, 2021-08-09:
The remote seems to remember the light used last. Maybe we can add call `self.light.on(Color.WHITE)` Remote.__init__() to get a clean state, which is consistent with re-connecting when re-starting the script.

_Originally posted by @laurensvalk in https://github.com/pybricks/support/issues/404#issuecomment-893413772_

dlech, 2021-08-09:
We could also make it match the "system" color of the hub. Although we haven't implemented a feature to change the hub color yet.

Vinz1911, 2021-08-09:
I discovered also a strange behavior. I have similar code to control, if I connect the remote to a Technic Hub the light is on and the color is white. if i connect the same remote to my prime hub, the light goes off after the remote is connected. the remote works without any problems. I can then also turn on the light again, so also no problems here. I don't know why this occurs, it's a bit weird 😆

closed 2021-08-19 by dlech