pybricks/support #428: Reintstall Mindstorms Inventor firmware after update by Spike Prime firmware

ChristophGaukel, 2021-08-11:
I bought a mindstorms inventor (Set 51515), then I downloaded the Spike Prime App and updated the firmware. Now the hub works well with the Spike Prime App, but no more with the Mindstorms Inventor App (it does not connect anymore).

I followed the instructions in https://dfu.pybricks.com (both in win10 and Linux), but the "Find Device" button does not find my device.

When I use terminal command lsusb (under Linux), then I see the device and it is in DFU mode:

 Bus 001 Device 009: ID 0694:0011 Lego Group LEGO Technic Large Hub in DFU Mode

What can I do, to reinstall the mindstorms inventor firmware?

dlech, 2021-08-11:
If you never installed Pybricks on this hub, then I would suggest getting in touch with LEGO customer support so that they are aware of the issue.

But you should be able to use our tools to help you get unstuck anyway. To be able to use the hub without root permission on Linux, you need to install [udev rules](https://github.com/pybricks/pybricksdev/blob/master/pybricksdev/resources/99-pybricksdev.rules) for it, then unplug it and plug it back in.

dlech, 2021-08-11:
Maybe there was a mistake in copying the rules? Or you didn't unplug/replug so that the rules take effect? I just tried it and it worked for me.

Also, I just published a [pbrick-rules](https://github.com/pybricks/pbrick-rules) Ubuntu package which may help (README has command for non-Ubuntu install too).

dlech, 2021-08-11:
And if all else fails, try rebooting everything.

dlech, 2021-08-11:
Yes. 

```shell
sudo apt update && sudo apt install dfu-util
dfu-util ??? (I will edit when I figure it out)
````

ChristophGaukel, 2021-08-11:
Thanks!
I did download file:
-rw-rw-r-- 1 christoph christoph 622592 Aug 11 18:21 /home/christoph/Downloads/firmware-backup-v1.2.01.0103-d08b6f.bin

terminal command "dfu-util -l" replies:
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [0694:0011] ver=0100, devnum=10, cfg=1, intf=0, path="1-1.2", alt=3, name="UNKNOWN", serial="3261385C3338"
Found DFU: [0694:0011] ver=0100, devnum=10, cfg=1, intf=0, path="1-1.2", alt=2, name="UNKNOWN", serial="3261385C3338"
Found DFU: [0694:0011] ver=0100, devnum=10, cfg=1, intf=0, path="1-1.2", alt=1, name="UNKNOWN", serial="3261385C3338"
Found DFU: [0694:0011] ver=0100, devnum=10, cfg=1, intf=0, path="1-1.2", alt=0, name="@LEGO LES HUB /0x08000000/02*016Ka,02*016Kg,01*064Kg,07*128Kg/0x10000000/01*1Ma/0x10100000/31*1Ma", serial="3261385C3338"

And I expect the command to load the firmware is: "dfu-util -d 0604:0011 -D /home/christoph/Downloads/firmware-backup-v1.2.01.0103-d08b6f.bin"

dlech, 2021-08-11:
See edited command above. There are additional required args.

ChristophGaukel, 2021-08-11:
thanks a lot!

command dfu-util  did successfully load the firmware and the Mindstorms Inventor App then did recognize the device and update the firmware to the newest version.

Here is the command line of dfu-util:
dfu-util --alt 0 --dfuse-addr 0x08008000 --download ~/Downloads/firmware-backup-v1.2.01.0103-d08b6f.bin

And here its output:
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0694:0011
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "LEGO LES HUB "
Downloading to address = 0x08008000, size = 622592
Download	[=========================] 100%       622592 bytes

laurensvalk, 2021-08-11:
I hope you don’t mind me asking a few more questions even though you’ve found a fix.

Did you install Chromium as a snap package or in another way?

Does it still produce the error if you enable the “enable-experimental-web-platform-features” flag in chrome://flags ?

ChristophGaukel, 2021-08-11:
My chromium was installed via snap, terminal command: "which chromium" replies:
/snap/bin/chromium

Enabling the “enable-experimental-web-platform-features” flag in chrome://flags does not change the behavior. As before I get:
SecurityError: Access denied.

laurensvalk, 2021-08-11:
Thanks for your response. This helps us improve those pages.

While it works fine for me on Ubuntu 18.04, I have the same issue on Ubuntu 20.04. On that laptop, it does work when installing Google Chrome.

I think we had something similar with Bluetooth support in the Chromium snap for a while, but that has since been fixed. Not sure if this is related.

ChristophGaukel, 2021-08-12:
My computer system is Ubuntu 21.04. I did install Google Chrome and indeed, the browser not only finds the device, it also connects and prints:

Name: LEGO Technic Large Hub in DFU Mode
Manufacturer: LEGO System A/S

Once again: thanks for your great support!

dlech, 2021-08-12:
Duplicate of #410?

laurensvalk, 2023-11-20:
This has gotten a bit better. Chromium as a snap now sees the device. But still gives this error:

[image]

laurensvalk, 2023-11-20:
https://bugs.launchpad.net/ubuntu/+source/chromium-browser/+bug/1905457

```
Version:

chromium 87.0.4280.66 1411

Steps to reproduce:

1. Have Chromium snap, with raw-usb plug disconnected.
2. Open Chromium and try to use a webpage that has webusb.

Result:

USB.getDevices() and USB.requestDevice() will succeed but USBDevice.open() will fail, emitting "DOMException: Permission Denied" on the browser console. Or, put another way, you can select a USB device that you want the webpage to use, but it is impossible for the webpage to actually use it.

Why this happens:

Chromium enumerates USB devices by querying udev. It does not require elevated permissions beyond the snap defaults in order to do this. So it can retrieve a list of USB devices even without raw-usb.

Impact:

It is impossible to access the USB devices until the raw-usb plug is connected. However, to a normal user, it appears that Chromium already has USB permissions because it can enumerate devices. The webusb API is in a confusing, partially-working state, and there is nothing to indicate that Chromium needs additional permissions in order for it to fully work.
```

dlech, 2023-11-20:
We had to make a formal request to get Bluetooth enabled by default: https://forum.snapcraft.io/t/request-to-autoconnect-chromium-to-bluez/18856

So maybe we can do the same for usb-raw?

laurensvalk, 2024-01-18:
> So we need to add that to: https://pybricks.com/projects/tutorials/dev/tools/linux/

Done as an intermediate solution via https://github.com/pybricks/pybricks-projects/commit/9ccd119331820f4f30eb040e970fb1616c42615d.

laurensvalk, 2026-10-11:
When Chromium is detected, we show this in the app now:

[image]

One task remaining here is to remove the outdated instructions from the website.