pybricks/support #432: [Bug] Remote.name() doesn't work on Technic hub

NStrijbosch, 2021-08-12:
**Describe the bug**
When running the sample script: [https://docs.pybricks.com/en/latest/pupdevices/remote.html#changing-the-name-of-the-remote](https://docs.pybricks.com/en/latest/pupdevices/remote.html#changing-the-name-of-the-remote) using the Technic hub the remote name is not printed (line 7) and the internal name in the remote is not changed. 

**To reproduce**
Steps to reproduce the behavior:
1. On a technic hub run the sample script: [https://docs.pybricks.com/en/latest/pupdevices/remote.html#changing-the-name-of-the-remote](https://docs.pybricks.com/en/latest/pupdevices/remote.html#changing-the-name-of-the-remote) 

**Expected behavior**
I expected the name to be printed and the internal name in the remote to be changed. On the PrimeHub this expected behaviour was obtained.

dlech, 2021-08-12:
Sending messages to LWP3 devices is currently very buggy. See #417. As a workaround, you can try adding `wait(500)` after every call.

SPIKE Prime works since it is using write with response instead of write without response. But I haven't had any luck yet making the same change to the City/Technic hubs.

dlech, 2021-08-17:
This should be fixed on the City and Technic hubs now. I still need to make similar changes on Move hub.

@NStrijbosch could you test and see if the Mario hack is still needed? The fixes for this may have fixed that too.

That is, delete this and see what happens:

```C
    // HACK: Characteristics of LEGO Mario are not properly found by GATT_DiscCharsByUUID().
    // remote_lwp3_char_handle for mario is hard coded for now
    if (context->hub_kind == LWP3_HUB_KIND_MARIO) {
        remote_lwp3_char_handle = 0x0012;
    }
```

NStrijbosch, 2021-08-18:
Yes, this specific bug is fixed now! 
From my first tests it seems sending messages from the technic hub to LWP3 devices (remote and city hub) is now more reliable. So this is a good step for multi hub!

On the other hand mario does have issues connecting with this update. I will investigate further!

dlech, 2021-08-18:
If Mario is raising a RutimeError during connection, it could be that we are getting a `ATT_EVENT_ERROR_RSP`, in which case the following change might work:

```diff
diff --git a/lib/pbio/drv/bluetooth/bluetooth_stm32_cc2640.c b/lib/pbio/drv/bluetooth/bluetooth_stm32_cc2640.c
index 92095995..6f94603a 100644
--- a/lib/pbio/drv/bluetooth/bluetooth_stm32_cc2640.c
+++ b/lib/pbio/drv/bluetooth/bluetooth_stm32_cc2640.c
@@ -554,6 +554,13 @@ try_again:
         HCI_StatusCodes_t status;
         (payload = get_vendor_event(remote_handle, &event, &status)) && ({
             if (event == ATT_EVENT_ERROR_RSP && payload[0] == ATT_READ_BY_TYPE_REQ) {
+                // HACK: Characteristics of LEGO Mario are not properly found by GATT_DiscCharsByUUID().
+                // remote_lwp3_char_handle for mario is hard coded for now
+                if (context->hub_kind == LWP3_HUB_KIND_MARIO) {
+                    remote_lwp3_char_handle = 0x0012;
+                    goto retry;
+                }
+
                 task->status = PBIO_ERROR_FAILED;
                 goto disconnect;
             }
@@ -577,12 +584,6 @@ try_again:
         });
     });
 
-    // HACK: Characteristics of LEGO Mario are not properly found by GATT_DiscCharsByUUID().
-    // remote_lwp3_char_handle for mario is hard coded for now
-    if (context->hub_kind == LWP3_HUB_KIND_MARIO) {
-        remote_lwp3_char_handle = 0x0012;
-    }
-
     // enable notifications
 
 retry:
```

dlech, 2021-08-18:
Fix for Move hub has been committed.

We can start a new issue for the Mario regression.

closed 2021-08-18 by dlech