pybricks/support #438: Flashing firmware on Android is slow

dlech, 2021-08-16:
**Describe the bug**
Flashing firmware on City Hub and Technic Hub using Android OS takes longer than on other platforms.

**To reproduce**
Steps to reproduce the behavior:
1. Get a stopwatch
2. Time firmware flashing on non-Android
3. Time firmware on Android

**Expected behavior**
Times should be reasonably close.

dlech, 2021-08-16:
As discussed in https://github.com/pybricks/support/issues/403#issuecomment-897897102, this is caused by a bug in Chromium where the MTU is negotiated to the minimum value on Android and higher values on other platforms.

Upstream bug is https://crbug.com/1164621. There has been a preliminary patch proposed there, so we just need to get it official submitted and merged.

dlech, 2021-11-16:
This has been fixed in Chromium 98.

We could modify https://github.com/pybricks/pybricks-code/blob/e7d44ea9a9195bc2a612a22acd84e09dfc048dac/src/firmware/sagas.ts#L382 to check for this version or just wait until most of the world has upgraded and remove the Android check there altogether.

dlech, 2022-12-08:
Fixed in https://github.com/pybricks/pybricks-code/commit/614e0b00743d2fb07d710078a28145336d04ab04

closed 2022-12-08 by dlech

laurensvalk, 2022-12-09:
What is the behavior on older phones after this fix? Just slow or broken?

dlech, 2022-12-09:
The age of the phone doesn't matter, only the version of Chrome. Firmware will fail to flash versions prior to 98 which was released over a year ago.

laurensvalk, 2022-12-09:
Ah. I was thinking of Android not getting updates on older phones., but indeed Chrome should still be getting updates on those.