pybricks/support #439: Support for SPIKE Essential hub (Technic Small Hub)

dlech, 2021-08-17:
**Is your feature request related to a problem? Please describe.**
LEGO Education just released a new 2-port hub.

[spec sheet](https://education.lego.com/v3/assets/blt293eea581807678a/blt8a69ecd244fee569/60fe864661f38746a5778f86/LE_SPIKE_Essential_Tech_fact_sheet_Small_Hub_45609_2HY21_Digital.pdf)

[image]

Assuming that this is similar enough to SPIKE Prime, we could potentially add support for both at the same time.

dlech, 2021-08-17:
Give this issue a :+1: if you would like to see support for this hub added sooner rather than later.

Vinz1911, 2021-08-18:
Awesome, this will be great with the upcoming Hub2Hub Feature. Why just 2 Ports, such a hub deserves 4, Damn Lego!!!!

dlech, 2021-09-14:
This is done. Please open new issues for any problems encountered.

closed 2021-09-15 by dlech