pybricks/support #440: Support for SPIKE Essential 3x3 Color Light Matrix

dlech, 2021-08-17:
**Is your feature request related to a problem? Please describe.**
LEGO Education just released a new Color Light Matrix I/O device with the new SPIKE Essential set.

[spec sheet](https://education.lego.com/v3/assets/blt293eea581807678a/blt8977c20602e50fd6/60fe864452fbad613927705b/LE_SPIKE_Essential_Tech_fact_sheet_Color_Light_Matrix_45608_2HY21_Digital.pdf)

[image]

dlech, 2021-08-17:
Give this issue a :+1: if you would like to see support for this hub added sooner rather than later.

laurensvalk, 2021-08-19:
We're going to need at least 6 :astonished: 

[image]

laurensvalk, 2021-09-08:
If anyone is curious, you can already try this device using the special build referenced above.

Then use this script to see what it could do.

```python

from pybricks.iodevices import PUPDevice
from pybricks.parameters import Port, Color
from pybricks.tools import wait

from urandom import randint

class ColorMatrix(PUPDevice):

    def _hue_to_id(self, h):

        h = h % 360

        # Color.RED
        if h < 15 or h > 330:
            return 9
        # Color.ORANGE
        if 15 <= h <= 45:
            return 8
        # Color.YELLOW
        if 45 <= h <= 90:
            return 7
        # Color.GREEN
        if 90 <= h <= 135:
            return 6
        # Turquoise?
        if 135 <= h <= 165:
            return 5
        # Color.CYAN
        if 165 <= h <= 210:
            return 4
        # Color.BLUE
        if 210 <= h <= 255:
            return 3
        # Color.VIOLET
        if 255 <= h <= 285:
            return 2
        # Color.MAGENTA
        if 285 <= h <= 330:
            return 1

    def _color_to_id(self, color):

        brightness = color.v // 10

        if color.s < 30:
            return 10 if color.v <= 10 else 10 + 16 * brightness
        else:
            return self._hue_to_id(color.h) + 16 * brightness

    def off(self):
        self.write(2, [0]*9)

    def on(self, colors):

        if type(colors) == type(Color.RED):
            self.write(2, [self._color_to_id(colors)] * 9)
        else:
            self.write(2, [self._color_to_id(c) for c in colors])

color_matrix = ColorMatrix(Port.B)

# Static single color example
color_matrix.on(Color.ORANGE)
wait(1000)

# Static multi color example
color_matrix.on([Color.RED, Color.WHITE, Color.BLUE] * 3)
wait(1000)

# Fade example
for i in range(3):
    for brightness in tuple(range(0, 110, 10)) + tuple(range(100, -10, -10)):
        color_matrix.on(Color.BLUE * (brightness / 100) )
        wait(50)

# Randomness
for i in range(1000):
    color_matrix.on([Color(randint(0, 360), v=40) for i in range(9)])
    wait(200)

```

laurensvalk, 2021-09-15:
This device seems quite susceptible to #304. And as pointed out in that issue, even without prints, possibly just a particular timing.

dlech, 2021-09-15:
Does the hub "freeze" or does it trigger the watchdog timer reset?

laurensvalk, 2021-09-15:
Watchdog.

dlech, 2021-09-17:
> This device seems quite susceptible to #304.

This is fixed now.

closed 2021-11-20 by dlech