pybricks/support #441: Support for SPIKE Essential Small Angular Motor

dlech, 2021-08-17:
**Is your feature request related to a problem? Please describe.**
LEGO Education just released a new Small Angular Motor with the new SPIKE Essential set.

[spec sheet](https://education.lego.com/v3/assets/blt293eea581807678a/blt20ee0f27f6735942/60fe86455483765886b0da3c/LE_SPIKE_Essential_Tech_fact_sheet_Small_Angular_Motor_45607_2HY21_Digital.pdf)

[image]

dlech, 2021-08-17:
Give this issue a :+1: if you would like to see support for this hub added sooner rather than later.

laurensvalk, 2021-11-22:
Documentation: Need to add this to the render with all the motors"

[image]

ZPhilo, 2021-11-22:
https://www.ldraw.org/cgi-bin/ptdetail.cgi?f=parts/68488c01.dat ;)

closed 2021-11-23 by laurensvalk