pybricks/support #442: Recent commit broke Mario hub connection

dlech, 2021-08-18:
On the other hand mario does have issues connecting with this update. I will investigate further!

_Originally posted by @NStrijbosch in https://github.com/pybricks/support/issues/432#issuecomment-901066531_

dlech, 2021-08-18:
https://github.com/pybricks/pybricks-micropython/commit/1235b35f148900db131bf946e54a383d2fca6460 added a hack to allow connecting to a Mario hub, however https://github.com/pybricks/pybricks-micropython/commit/d82175c9513eb227f2dbe465fc702daf040ca80a and https://github.com/pybricks/pybricks-micropython/commit/a4a85dd34adfcb419d4fefc40a8232da2defe710 broke the workaround while fixing other issues.

NStrijbosch, 2021-08-19:
Further observations while trying to connect to Mario from a Technic hub: 

It seems the technic hub gets stuck in the PT_WAIT_UNTIL from line: 550-578 in bluetooth_stm32_cc2640.c 
If I bypass these lines Mario does connect, but when attempting a write, i.e., pbdrv_bluetooth_write_remote, the program is stuck again.

NStrijbosch, 2021-08-30:
For me this is still an issue. 

I just tried (on a Technic Hub) the code you mentioned here: [https://github.com/pybricks/pybricks-micropython/pull/68](https://github.com/pybricks/pybricks-micropython/pull/68):
``` python
from pybricks.hubs import ThisHub
from pybricks.iodevices import LWP3Device
from pybricks.parameters import Color

MARIO = const(0x43)
PORT_VALUE_MSG = const(0x45)

hub = ThisHub()

# Connect to the Mario
mario = LWP3Device(MARIO)
print("connected")

#subscribe to Mario barcode/color sensor
mario.write(b'\x0a\x00\x41\x01\x00\x01\x00\x00\x00\x01')

while True:
    msg = mario.read()

    kind = msg[2]
    
    if kind != PORT_VALUE_MSG:
        continue

    color = msg[6]
    
    print("color:", color)

    if color == 21:  # Mario is on red
        hub.light.on(Color.RED)
    else:
        hub.light.on(Color.BLUE)
```

Mario seems to connect, but the line  `print("connected" )` is not executed. The technic hub freezes and needs a reboot. 

Since you mentioned you succeeded to run this example: could it be that our Mario's run different firmwares? I just updated the firmware of my Mario through the official app (So I am assuming my Mario is running the latest Firmware).  A firmware request through the LWP3 REPL yields the following result:
[image]

dlech, 2021-08-30:
I have the same Mario firmware version. I was testing using City hub (and Move hub).

Does the hub unfreeze if you disconnect Mario?

NStrijbosch, 2021-08-30:
Just tested using a city hub, that one does work... So it might be Technic hub specific (I didn't expect it to be different to the city hub)

> Does the hub unfreeze if you disconnect Mario?

No, the technic hub stays frozen when I shutdown Mario.

dlech, 2021-08-30:
>  So it might be Technic hub specific (I didn't expect it to be different to the city hub)

I didn't expect it to be different either. I will try the Technic hub later.

dlech, 2021-08-30:
Connecting is working for me with my Technic hub.

dlech, 2021-08-30:
Maybe one thing to check... with the official LEGO firmware, read the Radio Firmware Version hub property of the hub. This might tell us if we have different firmware on the Bluetooth chip between our two hubs.

NStrijbosch, 2021-08-30:
I can confirm it is this specific hub that is causing the issue. I just tested with another Technic hub which can run the program successfully.  

I flashed the official LEGO firmware on the first hub, but it is not recognized by the pybricksdev LWP3 REPL. Which is strange in itself. But also prevents me from checking the radio firmware version at the moment. I can investigate this specific hub a bit more in the coming days. 

Just as an FYI:
I used this specific technichub over the past couple of weeks for my tests so:
- before this commit it could connect to Mario and read/write 
- at the moment it can still connect to a City hub

dlech, 2021-08-30:
Very interesting. Let's start a new issue for this specific hub. Since it doesn't work with Pybricksdev either, it is not just a Mario thing.