pybricks/support #445: Mario class for connecting to Mario/Luigi via Bluetooth

dlech, 2021-08-20:
**Is your feature request related to a problem? Please describe.**
This stems from the discussion in https://github.com/pybricks/pybricks-micropython/pull/68 where @NStrijbosch  has proposed a generic LWP3 class. While this is certainly useful and stands on its own, it does require that users encode and decode messages from the LWP3 protocol which is not exactly beginner-friendly.

**Describe the solution you'd like**
Since Mario seems to be a somewhat simple device (no motors, no I/O ports), it seems like we could make a specialized `Mario` class that provides a better API for using Mario like a sensor.

**Describe alternatives you've considered**
Just use the generic LWP3 class for Mario.

**Additional context**
I don't have a Mario hub, so I don't know what all of the capabilities are yet. Is there a single configuration that would work for 80% of users?

NStrijbosch, 2021-08-21:
> I don't know what all of the capabilities are yet

As far as I know Mario has 3 sensors:
*  The barcode/color sensor. With the modes:
        -  recognize both barcodes an ID's of colors
        -  RGB mode (i am not sure if it is possible to combine this with barcode scanning in a combi mode)
*  The IMU. WIth the modes:
        -  acceleration
        -  gestures, e.g., freefall (There are a whole bunch more than the standard gestures in the Prime hub)
* The Pants sensor. With the mode
        - Detect what kind of pants Mario/Luigi is wearing
        
Some other relevant stuff that you can do:
* Turn up/down its volume

> Is there a single configuration that would work for 80% of users?

I would say this should include:
*  Barcode/Color ID's (I think including the barcodes is important, this is the only LEGO sensor with this capability, therefore a reason to use Mario/Luigi)
*  Gesture 
*  Pants sensor 
*  Volume control

Tcm0, 2021-09-15:
https://github.com/bricklife/LEGO-Mario-Reveng/ contains some information on the barcodes, the modes of the hub etc