pybricks/support #461: [Feature] USB OTG (on the go) on spike prime/robot inventor hub USB port

k82k8, 2021-08-29:
Hi,

since I have expressed my wish for Lego to add USB OTG for some time now, to no avail so far, I decided to see if the pybricks firmware could be used as a means to do this. I have researched the processor of the spike prime/robot inventor hub and according to the spec sheet, it holds an ATMEL microcontroller:

quoted from : https://education.lego.com/v3/assets/blt293eea581807678a/bltf512a371e82f6420/5f8801baf4f4cf0fa39d2feb/techspecs_techniclargehub.pdf

 "The Hub is powered by a 100MHz M4 320 KB RAM 1M FLASH processor"

I searched a while ago for the exact processor, and it is an ATMEL M4, then when I checked the data sheet for it, it does indeed support USB OTG.

I am not sure what the upcoming spike prime small 2 port hub will contain, processor-wise, but it would be cool if we could control USB peripherals like webcams, USB sticks, etc from the spike prime/robot inventor hub.

laurensvalk, 2021-08-29:
The processor is a STM32F413. 

See also: https://bricks.stackexchange.com/questions/15057/what-internal-electronics-do-spike-prime-and-mindstorms-robot-inventor-use

dlech, 2021-08-30:
If the Technic Large hub is like the EV3, then the port can only be used in peripheral mode. On EV3, there are two problems: 1) there is no 5V output wired to the port and 2) the ID pin of the USB port is not attached to the CPU.

From the picture, we can see that there are only 4 wires coming from the USB port, so we can tell that the Technic Large hub has at least problem 2) the ID pin is not attached to anything. So it won't be possible to do proper OTG to allow both host and peripheral devices (at least not without soldering an extra wire on).

If the required electronics are in place, maybe it would be possible to operate the USB port in host-only mode though? I'm not sure how this would effect charging the batter either since the USB port is used for that.

dlech, 2023-04-28:
As mentioned already, this is not technically possible without modifying the hardware, so this is not something we will be supporting and leave it to the serious hackers.

closed 2023-04-28 by dlech