pybricks/support #464: [Bug] unable to pair City Hubs in Code screen

jjsherm2, 2021-09-02:
**Describe the bug**
City Hubs won't connect when trying to download Pybrick firmware.  Error message displayed: "Unexpeced error while trying to connect:  Connection failed for unknown reason."

Same issue exist for two different City Hubs trying to connect to an LG Gram laptop running windows 10 and on a Samsung Galaxy Tab 5Se.

dlech, 2021-09-02:
Please see https://github.com/pybricks/support/discussions/270. Will will need more information, like logs, to find the source of the problem.

jjsherm2, 2021-09-02:
> Also, please try https://beta.pybricks.com to see if the issue has already been fixed.

The beta site has the same issue.

John Sherman

jjsherm2, 2021-09-02:
Not helpful, the only thing that applied was to reboot windows and that didn't work and there was nothing related to Samsung devices.  I am more than happy to provide additional info if you can tell me what you need and how to find it (for example: I don't know what logs you would be looking for or where they are).  Please forgive my ignorance I haven't worked with programming for many years.

Thanks
John Sherman

dlech, 2021-09-02:
The troubleshooting guide at #270 describes what we are looking for. Are the errors in the "Developer Console"? What is in the "Chrome Device Log"? If none of those have anything to identify the problem, then "Capturing Bluetooth Packets".

jjsherm2, 2021-09-03:
David thanks for the clarification.  attached are screen images of the Developer Console and Chrome Device Log.  I haven't been able to try the Packet capture.
Also note that on an attempt to after these logs, got the same error messages but the City Hub light started flashing red, blue, green in a continuous cycle.  It was not connected and would not leave this mode until the battery was pulled out.
thanks,
John Sherman

jjsherm2, 2021-09-03:
[Issue#464.pptx](https://github.com/pybricks/support/files/7106812/Issue.464.pptx)
Here is the file.

jjsherm2, 2021-09-03:
I have now successfully flashed both my hubs.  The issue may have been I just didn't hold the button on the hubs long enough.  You might want to consider rewording the instructions to emphasize this.  My apologies for taking up your time.
Thanks,

dlech, 2021-09-03:
No problem. This is valuable feedback that can help us improve so we are glad you took the time to figure out what the problem was and let us know about it instead of giving up right away and not giving any feedback.

dlech, 2021-09-03:
In addition to documenting it, we are also considering improving the firmware update process to not require users to manually start the hub in update mode, in which case holding the button should not be needed at all.

dlech, 2021-09-03:
There is also an existing issue about improving the error messages if a hub disconnects while flashing firmware: https://github.com/pybricks/pybricks-code/issues/255

dlech, 2022-07-28:
The UI has been improved in Pybricks Code v3.2.0b3 (Pybricks Beta v2.0.0-beta.3).

closed 2022-07-28 by dlech