pybricks/support #465: [Feature] File browser in Pybricks Code

laurensvalk, 2021-09-03:
**Is your feature request related to a problem? Please describe.**

A file browser could potentially capture a lot of previously requested features and use cases:
- Having just one program can be frustrating. It's easy to lose programs by accidentally overwriting them.
- Multi-file projects and using libraries (SPIKE/MINDSTORMS)
- Now you can have multiple "tabs", by just having Pybricks Code open in multiple browser tabs.
- Multi-hub coding. As above: just connect one browser tab to one hub.
- We could open example projects directly in Pybricks Code with a special URL. This action would just add that file.

If we want anything more fancy than this, we should perhaps look at making a Visual Studio Code extension, but if a file browser isn't too hard, this could work well for 95% of cases, including fairly advanced ones.

**Describe the solution you'd like**
A file browser like this. Ideally we can use something that exists already. The existing file buttons can be dropped.

[image]

**Describe alternatives you've considered**
Use Pybricksdev.

**Additional context**
Related: #173, #418, #84

dlech, 2021-09-03:
> Multi-file projects and using libraries (SPIKE/MINDSTORMS)

I think we should first have a bit more of think about what is a "project" and what is a "library". 

Does a project have a single main program? Is it a collection of various main programs for a single robot? Can a project target multiple hub types?

Is a library a collection of common functions for a single robot? Is a library a collection of generic functions to be share between multiple projects? Can hubs without external flash memory use libraries?

What specific problems are we trying to solve (user story)? 

For example, I would like to make a generic library with generator functions to provide async functionality for motors, timers, etc. This library should be globally available to all programs/projects. Then I might like to create a common Robot class for a specific LEGO model build (this would use the global generic library). Then I would write multiple programs using the Robot class for different activities.

Then I would like to easily share this with others. In other words, we should be able to package all of this in to a single (.zip?) file with a single click that can then be imported by another user, again with a single click. Can we do this without having duplicate copies of the generic library in each project folder?

Is this too advanced of a use case for out target audience (people who know little to no coding)?

Another use case would be for when we get hub-to-hub communication. We would want two copies (windows/browser tabs) of Pybricks Code open with different programs/projects open connected to two different hubs.

Are users actually going to create projects so big that it makes sense to break it up into multiple files or is it just about code reuse in multiple programs?

Can files and folder just be an implementation detail? 

It could be easier for non-technical people to just have "programs", "libraries" and "projects" without having to have any knowledge of a file system or having `.py` files be a program and/or a library. And we may not even be using a real file system, so this could throw off users who might expect to be able to find the virtual files and folders on their hard drive.

What are some other ways to solve these problems?

Idea: instead of having "libraries", we allow users to create a collection of code snippets. These snippets could then be inserted into a single file program. For advanced users that care about code reuse, we could have a specialized comment (e.g. `# snippet: name`) that surrounds a snippet that has been inserted. Then if the source snippet is edited, the places where it is used could show an autocorrect light bulb that allows the user to chose to update to the new version of the snippet if they want to. It could also work the other way around, if they update the snippet in the program, the light bulb will suggest updating the source snippet.

laurensvalk, 2021-09-03:
What constitutes a project is worth a dedicated (probably separate) discussion.

Almost all variants are going to require some form of file explorer, so it could make sense to start thinking about it.

laurensvalk, 2021-09-10:
To simplify this, perhaps we can consider just a file list to begin with:

This already has most of the aforementioned benefits, and we'd just save multi-file concepts for a future version.

[image]

laurensvalk, 2021-09-10:
To make it an even more minimal change, it could be done this way:

- Keep the open button, but show as an "upload icon"
- Keep the save button, but show as a "download button"
- Add one delete button

[image]

Thanks @afarago for discussion & ideas.

dlech, 2021-09-10:
Can we have a toolbar button to toggle this sidebar so that it doesn't have to be open all of the time?

laurensvalk, 2021-09-10:
Yes, perhaps the file icon (deleted in screenshot above) could be used for that purpose.

dlech, 2022-06-03:
This feature has been released on <https://beta.pybricks.com>.

closed 2022-06-03 by dlech