pybricks/support #466: [Bug] "Long" Handset connection process if Primehub is not connected to Console

Vinz1911, 2021-09-05:
**Describe the bug**
I'm not sure if this is a bug. I discovered that the handset took longer to connect if I store a program on the Hub without attached to the console. It's not related if the program is stored. if I connect the hub to the Pybricks website and run the stored program that flashed with the firmware. the connection process is as fast as expected. 

what means "long". it took about 6,5 seconds to connect to the remote if the console is not attached. 
if the console is attached, it took only about 2 seconds.

I only discovered this with the Spike/RI Hub. On the Technic Hub this "problem" does not exist

**To reproduce**
Steps to reproduce the behavior:
1. make a simple program which connects to the remote
2. put it in the main.py from the firmware
3. flash the firmware
4. run the program (without connected to Pybricks website)

**Expected behavior**
same connection time with or without attached console

dlech, 2023-02-24:
I am able to reproduce using firmware v3.2.2-456-g830579b2 and the following program. 

```python
from pybricks.hubs import PrimeHub
from pybricks.pupdevices import Motor, ColorSensor, UltrasonicSensor, ForceSensor, Remote
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch

hub = PrimeHub()

# turn on remote before starting program for accruate time!

connect_timer = StopWatch()
remote = Remote()
hub.display.text(f"{connect_timer.time() / 1000:0.3f}")
```

dlech, 2023-02-24:
Sniffing packets shows the delay is after enabling notifications on the LWP3 hub characteristic. Eventually a *Connection Parameter Update Request* comes from the remote and gets things moving again.

[image]

laurensvalk, 2023-02-25:
I'm re-opening this momentarily as a reminder to double check that this also fixes the delay of connecting to an LWP3 Duplo device. It sounds like it should.

dlech, 2023-02-27:
Confirmed that `LWP3Device` is fixed as well (using mario).

closed 2023-02-27 by dlech