pybricks/support #471: #!/usr/bin/env pybricks-micropython returns error message in visual studio code

FinnThom, 2021-09-09:
I am trying to use Visual Studio Code to write python code for the Lego Mindstorms EV3.

I followed the [official guide](https://le-www-live-s.legocdn.com/sc/media/files/ev3-micropython/getting-started-with-micropython-v2_enau-0e0a84a90ce27550aa995395695e687d.pdf), though I already had VS Code. The EV3 micropython extension is installed and enabled. The EV3 is connected and the microSD card is inserted.

[image]

laurensvalk, 2021-09-09:
From the screenshot, it looks like Visual Studio Code is trying to run your script on your *computer* instead of on the *robot*. 

To find out why this happens, can you check which debug configuration is active? It should look like this:

[image]

FinnThom, 2021-09-09:
Thanks for the answer. I hope this is what you meant.

[image]

laurensvalk, 2021-09-09:
What happens if you go to the debug tab in the menu on the left and click on the green triangle?

[image]

FinnThom, 2021-09-09:
[image]

closed 2023-03-21 by dlech