pybricks/support #472: #!/usr/bin/env pybricks-micropython returns error message in VS Code

FinnThom, 2021-09-09:
I am trying to use Visual Studio Code to write python code for the Lego Mindstorms EV3.

I followed the [official guide](https://le-www-live-s.legocdn.com/sc/media/files/ev3-micropython/getting-started-with-micropython-v2_enau-0e0a84a90ce27550aa995395695e687d.pdf), though I already had VS Code. The EV3 micropython extension is installed and enabled. The EV3 is connected and the microSD card is inserted.

[image]

laurensvalk, 2021-09-09:
From the screenshot, it looks like Visual Studio Code is trying to run your script on your *computer* instead of on the *robot*. 

To find out why this happens, can you check which debug configuration is active? It should look like this:

[image]

dlech, 2021-09-09:
It looks like the discussion continued in #471, so we'll close this duplicate issue.

closed 2021-09-09 by dlech