pybricks/support #473: [Bug] SPIKE Prime Color sensor problem

afarago, 2021-09-09:
**Describe the bug**
spike color sensor is initialized and the code is stopped the color sensor turns dark and does not respond anymore.
sensor unplug-replug and then app restart is required.

**To reproduce**
Steps to reproduce the behavior:
1. goto beta.pybricks.com
2. paste sample code
from pybricks.hubs import PrimeHub
from pybricks.pupdevices import Motor, ColorSensor, ForceSensor, ColorDistanceSensor, InfraredSensor, TiltSensor
from pybricks.parameters import Port, Color, Direction, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch
from pybricks.geometry import Matrix
from pybricks.iodevices import PUPDevice
from math import fabs
hub = PrimeHub()
propeller_motor=Motor(Port.F)
#"sun" colorsensor
nap_colorsensor=ColorSensor(Port.E)
#"switch" function motor
kapcsolo_motor=Motor(Port.D)
libra_forcesensor = ForceSensor(Port.B) 

while True:
    color = nap_colorsensor.color()
    kapcsolo_allas=kapcsolo_motor.angle()
    vizszint=libra_forcesensor.force()
    
    if color == Color.YELLOW and kapcsolo_allas<30 and vizszint<1:
        propeller_motor.run(100)
    else:
        propeller_motor.stop()
    #print(vizszint)
    if color != Color.YELLOW and not kapcsolo_allas<30 and vizszint>0.2:
        hub.speaker.play_notes(['C4/4','E4/4','C4/4','E4/4'],300)

(app functionality for https://youtu.be/Bge3OQpJX7c?t=194)

**Expected behavior**
What did you expect to happen instead?
app runs smoothly several times.
instead the app is started and is working fine, yet after stopping the app the color sensor turns dark
at next run the color sensor value reading is not working anymore

**Screenshots**
n/a

laurensvalk, 2021-09-09:
Thanks for reporting. This was fixed in https://github.com/pybricks/support/issues/456

I think we can probably update the version that's included in the installer. A lot has already been updated in just the last two weeks since the previous release.

laurensvalk, 2021-09-09:
The installer has been updated to the latest version. You may need to hard refresh using `CTRL+F5` before you install, to ensure you get the latest version.

To confirm, run this:

```python
from pybricks import version
print(version)
```

This should give:

```
('primehub', '3.1.0a4', 'v3.1.0a4-14-ge934075d on 2021-09-09')
```

afarago, 2021-09-09:
Thanks for the information and the quick update.

Several options I have tried to get the updated fw:
Reinstalled the pybricks fw; reset then reinstalled; hard reloaded installer page with ctrl-F5 and reinstalled; tried incognito mode

Whatever I did - I still received version ('primehub', '3.1.0a3', 'v3.1.0a3-62-g51bdd097 on 2021-08-26') 
firmware-dual-boot-base.bin file is 235484 bytes long.

Will probably retry in a few days.

laurensvalk, 2021-09-09:
Try this:

- Press `F12` to open the debug menu. (But you don't have to actually use it.)
- Right click on the refresh button and pick the third option:

[image]

afarago, 2021-09-09:
Thank you so much! That did the trick to upgrade to ('primehub', '3.1.0a4', 'v3.1.0a4-14-ge934075d on 2021-09-09').

On that version the original bug for the color sensor is indeed solved, I have checked with reported code.

Thanks for the great work!

closed 2021-09-09 by afarago