pybricks/support #480: [Bug] Can only use one sensor on Essential Hub

laurensvalk, 2021-09-15:
**Describe the bug**
At the moment, you can use only one UART sensor on the Essential Hub. The other one will stop responding with no data since the last keep-alive message.

Given that the [motor ports A an B were flipped](https://github.com/pybricks/pybricks-micropython/commit/af6c48995a57322bac485f64795d2f3b5b1e25b4), I suppose it's possible some of the auto-id pins are swapped or duplicated too :smile: 

**To reproduce**
Try to use two sensors or motors. It's even easier to see with the Color Distance Sensor since you can see it going out of sync.

**Expected behavior**
Both ports just work.

laurensvalk, 2021-09-15:
> Does a single motor/sensor work on each port?

Yes, it looks that way.

laurensvalk, 2021-09-15:
I'll try making a reproducable case later, to rule out any new issues related to the Color Light Matrix.

laurensvalk, 2021-09-15:
Reproducable cases.`uartdev` debug print is *on*:

Boot with everything unplugged. Connect via Pybricks code to see debug messages.

**1**

Plug in Color and Distance Sensor to port A:
- Correct debug output.
- Light stays in color mode, so is synced up.

Plug in Color and Distance Sensor to port B:
- Does not sync up and gives no debug output.

**2**
As above. But sometimes when adding the second sensor, *both* fall out of sync.

**3**
As above, but sometimes they *do* sync up.

laurensvalk, 2021-09-15:
Hmm, I might be on to something in a new experiment. I have something reproducible. Will report back asap.

laurensvalk, 2021-09-15:
```diff
diff --git a/pybricks/hubs/pb_type_essentialhub.c b/pybricks/hubs/pb_type_essentialhub.c
index d8e65ff4..4056d320 100644
--- a/pybricks/hubs/pb_type_essentialhub.c
+++ b/pybricks/hubs/pb_type_essentialhub.c
@@ -42,7 +42,11 @@ STATIC mp_obj_t hubs_EssentialHub_make_new(const mp_obj_type_t *type, size_t n_a
     hubs_EssentialHub_obj_t *self = m_new_obj(hubs_EssentialHub_obj_t);
     self->base.type = (mp_obj_type_t *)type;
     self->buttons = pb_type_Keypad_obj_new(MP_ARRAY_SIZE(essentialhub_buttons), essentialhub_buttons, pbio_button_is_pressed);
-    self->imu = pb_type_IMU_obj_new(top_side_in, front_side_in);
+
+    (void) top_side_in;
+    (void) front_side_in;
+
+    // self->imu = pb_type_IMU_obj_new(top_side_in, front_side_in);
     self->light = common_ColorLight_internal_obj_new(pbsys_status_light);
     return MP_OBJ_FROM_PTR(self);
 }
```

This works around it.

Sensors stopped as soon as `EssentialHub` was initialized.

That still doesn't explain the color distance sensor sometimes going out of sync, but at least this happened consistently.

dlech, 2021-09-15:
Just pushed a commit to fix the IMU pin mux issue.

laurensvalk, 2021-09-15:
This fixes it indeed. Thanks!

closed 2021-09-15 by laurensvalk