pybricks/support #481: [Bug] "Can't import name Remote" wtf?

piotrmil, 2021-09-15:
I was really excited to try this. I bootloaded the program onto my Technic hub, copied the defeault program, clicked the play button, and....

error. 

Apparently on line 1. Can't import name Remote.

So, wtf?

laurensvalk, 2021-09-15:
The remote feature is still in beta. It looks like you are using the stable version, which does not have the remote yet.

If you install the firmware and run the program with this version, it should work:

https://beta.pybricks.com/

piotrmil, 2021-09-15:
Cool. I'd suggest adding that to the error code to avoid confusion.

piotrmil, 2021-09-15:
Wait, I AM using beta. what is going on?!

laurensvalk, 2021-09-15:
```python
from pybricks import version
print(version)
```

What is the output if you run this?

piotrmil, 2021-09-15:
technichub 3.0.0 8845360 on 2021-06-09

laurensvalk, 2021-09-15:
Is that all the output you see? Could you try a screenshot instead? It seems as if a few characters are missing.

piotrmil, 2021-09-15:
I'm using it on a different computer, I have to go back and forth, what AM I missing apparently?

dlech, 2021-09-15:
> 2021-06-09

This is not the latest firmware. Please update the firmware using the firmware update button on https://beta.pybricks.com.

piotrmil, 2021-09-15:
I clicked "check for update" and it says that it is up to date. Yes, I am on the beta. page.

dlech, 2021-09-15:
[image]

this button

laurensvalk, 2021-09-15:
But you might have installed the firmware on the hub using the older version.

We probably have to add an indication that a firmware update is required.

dlech, 2021-09-15:
> We probably have to add an indication that a firmware update is required.

pretty sure we already have this

laurensvalk, 2021-09-15:
I'll give it a try. It seems like this question about the remote has come up a lot.

piotrmil, 2021-09-15:
The Firmware button is greyed out when I pair the device. Can't click it.

piotrmil, 2021-09-15:
> 
> 
> But you might have installed the firmware on the hub using the older version.
> 
> We probably have to add an indication that a firmware update is required.

Yeah, that sounds sensible I'd say.

laurensvalk, 2021-09-15:
You install the new firmware in the same way you did it before.

That is, turn the hub off. Then press and hold the button until it blinks purple. Then click the installation button.

dlech, 2021-09-15:
there are instructions and a video here: https://pybricks.com/install/technic-boost-city/

dlech, 2021-09-15:
> I'll give it a try. It seems like this question about the remote has come up a lot.

This is the expected message: https://github.com/pybricks/pybricks-code/blob/094807ccb513b3bb1908b95bc45214e3bb0557a8/src/notifications/i18n.en.json#L9

This message also happens in response to BlueZ bugs. So we could probably add a better version check and message.

piotrmil, 2021-09-15:
Okay, now it works like a charm :) version 3.1.0a4 from 2021-08-30.

Thanks! I would suggest expanding error messages to give some instructions to avoid giving you guys more work here ;) Especially now since RacingBricks made a video about you.

dlech, 2021-09-15:
What OS are you using Pybricks Code on?

dlech, 2021-09-15:
We started a new issue for the informational message, so closing this issue since the problem is resolved.

closed 2021-09-15 by dlech