pybricks/support #482: Recommend firmware update in Pybricks Code

laurensvalk, 2021-09-15:
If the firmware on the connected hub is older than the one provided by the running Pybricks Code release, perhaps we can add a notification like this:

"A new firmware version is available for this hub. Please install the latest version to use all new features."

And we could link to installation instructions.

_Originally posted by @laurensvalk in https://github.com/pybricks/support/issues/481#issuecomment-920283158_

laurensvalk, 2021-09-15:
> pretty sure we already have this

_Originally posted by @dlech in https://github.com/pybricks/support/issues/481#issuecomment-920284900_

laurensvalk, 2021-09-15:
> pretty sure we already have this

When installing the latest stable firmware, `('technichub', '3.0.0', '8845360 on 2021-06-08')`, Pybricks Code Beta does not seem to give a notification when I connect.

dlech, 2021-09-15:
In v3.1, we added the BLE Device Information service to have a good way to get the current firmware version of the hub. However, v3.0 doesn't have this service. The idea was that the absence of this service would be good enough to detect v3.0, but it turns out there are also BlueZ bugs that can trigger the same missing service even though the firmware has been updated.

laurensvalk, 2021-09-15:
Got it, thanks. So this issue should slowly disappear when we release V3.1.

dlech, 2021-09-15:
> When installing the latest stable firmware, `('technichub', '3.0.0', '8845360 on 2021-06-08')`, Pybricks Code Beta does not seem to give a notification when I connect.

If you are testing on Linux, you need to work around BlueZ bugs by removing the device from BlueZ and manually deleting the BlueZ cache for that device. See https://github.com/pybricks/support/discussions/270.

dlech, 2021-09-15:
> In v3.1, we added the BLE Device Information service to have a good way to get the current firmware version of the hub. However, v3.0 doesn't have this service.

I did some more digging and my memory is faulty. The Device Information Service did make it into 3.0 (https://github.com/pybricks/pybricks-micropython/commit/3b1b2d61676e767a72fc1e2b18b48cd944f393eb).

So we do need to fix this in Pybricks Code.

closed 2021-09-15 by dlech