pybricks/support #484: [Question] Strange behaviour on Motor.run with acceleration limit

vascolp, 2021-09-16:
Hi,

beta.pybricks.com loaded yesterday 2021-09-15, on a Technic hub, Technic Large or Extra Motor connected to any port.
Program:
    from pybricks.pupdevices import Motor
    from pybricks.parameters import Port
    from pybricks.tools import wait
    
    m= Motor(Port.B)
    m.control.limits(acceleration=35791) 
    m.run(0)
    wait(1000)
    print('Bye')

This program does nothing, as expected. 
However, if instead of 35791 you use 35792 or above, the motor will rotate at maximum speed during the wait....
Can you explain?

Thanks and Regards,
Continue th egood work!
Vasco

laurensvalk, 2021-09-17:
Thanks for opening this issue! This is a bug.

> Can you explain?

Everything above 32767 is probably undefined behavior. This is due to the way we currently convert user input values before saving the motor settings. This is done with a library called `libfixmath`, which stores decimal numbers up to 32767 only.

We should probably add more checks to deal with large input numbers, or reconsider the use of this library.

We avoided certain checks in some of the more advanced settings until now, to save space on the hub.

vascolp, 2021-09-17:
Thank you for your explanation, I thought it would be something like that.
I tried high numbers for the acceleration because I could not find anything in the documentation about the limits of this parameter... I want the motors to work like PF motors, that is, start immediately at maximum speed.... what should I use?
I understand very well space the space saving issues, I intend to do a “big” program and that matters!
By the way, where can I know the space limits of each hub, how much space my program is using, and things like that?

laurensvalk, 2021-09-17:
If you just want the motors to run without speed control (like PF 1.0), you can use the [`Motor.dc`](https://docs.pybricks.com/en/stable/pupdevices/motor.html#pybricks.pupdevices.Motor.dc) method.

That's still not going to give you an acceleration of over 32767 degrees per second squared, but it's the fastest it can do :rocket: 

> By the way, where can I know the space limits of each hub, how much space my program is using, and things like that?

You can read some tips and references to the documentation here: #483 .

closed 2022-12-06 by laurensvalk