pybricks/support #485: [Bug]  Remote(timeout = None) only connects in the first 10 seconds with TechnicHub and CityHub

NStrijbosch, 2021-09-19:
**Describe the bug**
When trying to connect to the remote using a TechnicHub or a CityHub setting the timeout larger than 10000 ms does not lead to the desired behaviour. The remote will only connect in the first 10 seconds after initializing the remote. When pressing the button on the remote more then 10 seconds after this, will never lead to a connect, however the program will wait for the timeout. 

**To reproduce**
Steps to reproduce the behavior:
1. On a TechnicHub/CityHub run the program
``` python
from pybricks.pupdevices import Remote

# Connect to any remote. Search forever until we find one.
my_remote = Remote(timeout=None)
print('connected')
```
2. Wait more than 10 seconds after starting the program
3. Press the button on the remote to attempt a connect

**Expected behavior**
The remote to connect after 10 seconds when the timeout is set larger then 10000.

NStrijbosch, 2021-09-19:
I expect the following cause:
The line 434 in [lib/pbio/drv/bluetooth/bluetooth_stm32_cc2640.c](https://github.com/pybricks/pybricks-micropython/blob/master/lib/pbio/drv/bluetooth/bluetooth_stm32_cc2640.c#:~:text=GAP_DeviceDiscoveryRequest(GAP_DEVICE_DISCOVERY_MODE_ALL%2C%201%2C%20GAP_FILTER_POLICY_SCAN_ANY_CONNECT_ANY)%3B)
```
GAP_DeviceDiscoveryRequest(GAP_DEVICE_DISCOVERY_MODE_ALL, 1, GAP_FILTER_POLICY_SCAN_ANY_CONNECT_ANY);
```
Does start a scan that automatically stops after 10 seconds. After 10 seconds the gap event `GAP_DEVICE_DISCOVERY_DONE` is called. Which could be used to start a new scan when timeout > 100000.

dlech, 2021-09-19:
Nice find. We could either fix it like you suggest or consider removing the timeout argument and just make a fixed 10 second timeout (if people _really_ need to scan longer than that, they could catch the timeout exception and try again).

NStrijbosch, 2021-09-19:
For the remote the fixed 10 second seems fine to me.

I stumbled upon this issue when trying to receive broadcasted data in the advertisement data from a robot inventor hub, which turned out to work only the first 10 seconds. In that case I _really_ want to scan indefinitely in the background. I actually was hoping for a cleaner solution then restarting a scan every 10 seconds.

dlech, 2021-09-19:
It looks like there is `GAP_SetParamValue(TGAP_LIM_DISC_SCAN, duration)`, but `duration` is 0 to 65535, so unless one of the values (e.g. 0) is a "magic" value that means infinite, I think we are going to just have to keep restarting.

dlech, 2021-09-19:
https://dev.ti.com/tirex/explore/node?node=AD4sGbaamTCyn0DvZgBAsg__krol.2c__LATEST indicates 0 might work

NStrijbosch, 2021-09-20:
Thank you for the useful resources! 

Adding the following lines:
``` c
    PT_WAIT_WHILE(pt, write_xfer_size);
    GAP_SetParamValue(TGAP_GEN_DISC_SCAN, 0);
    PT_WAIT_UNTIL(pt, hci_command_status);
    // ignoring response data

    PT_WAIT_WHILE(pt, write_xfer_size);
    GAP_SetParamValue(TGAP_LIM_DISC_SCAN, 0);
    PT_WAIT_UNTIL(pt, hci_command_status);
    // ignoring response data
```
extends the scanning time to 'forever'  (I tested until my patient ran out, so about 60 seconds ;) )

dlech, 2021-09-20:
Awesome. I'll take a PR for this change along with some comments explaining that this sets an infinite timeout and that we implement our own timeout.

closed 2021-11-17 by dlech