pybricks/support #498: [Question] How to flash a specific (CI) firmware onto a primehub

BertLindeman, 2021-09-29:
**Question**
Where can I find the current doc about how to flash CI-firmware onto a primehub

**Context**
(Sorry, my memory..)
I know it is not yet really supported, but I am very curious about the filesystem on the hubs.
I want to try [this snippet](https://github.com/pybricks/pybricks-micropython/commit/7ca4742d47c091103946aeb5227abc6a56976d5d)
Looking at the update, I think this experiment is only available on the primehub.
Using [install](install.pybricks.com) does not install the most current CI-firmware.

**Screenshots**

dlech, 2021-09-29:
pybricksdev flash firmware.zip

BertLindeman, 2021-09-29:
What I forgot was to keep pressing the bluetooth button before connecting with the USB cable...
Thank you David. The hint triggered enough to get going.

laurensvalk, 2021-10-06:
> I want to try this snippet

Since you ask --- If you want to try something even more experimental, try [this build](https://github.com/pybricks/pybricks-micropython/suites/3975276573/artifacts/99973427).

Try running a new program. Then try pressing the center button to see what happens. :wink:

laurensvalk, 2021-10-07:
Are you sure that is the version I linked to above? It should be from October 6.

BertLindeman, 2021-10-07:
Ah, right.

So run **a** program.

And on the center press after the run . . . 
That same program runs again.
Sooo, the program (probably main.py) **is** stored onto the hub.

Side step:
Am I right that the experimental file classes are not included here (or renamed?)
```
from pybricks.experimental import flash_init_spi, flash_init_littlefs, flash_read_raw, flash_read_file
```

results in
```
Traceback (most recent call last):
  File "main.py", line 5, in <module>
ImportError: can't import name flash_init_spi
```

BertLindeman, 2021-10-07:
> To see what's included, just enter the REPL, begin typing `flash` and hit tab to auto-complete.

Of course. 

So file access is becoming simpler, no init needed now.

My snippet:
``` python
# from pybricks.experimental import flash_init_spi
# from pybricks.experimental import  flash_init_littlefs
from pybricks.experimental import  flash_read_raw
from pybricks.experimental import  flash_read_file

# Init SPI and file system
# flash_init_spi()
# flash_init_littlefs()

# Read information about the file system
info = flash_read_raw(address=0, len=58)
print(info)

# Read an existing file. Use the original firmware to create this file.
filename = "main.py"
data = flash_read_file(filename)
print("\nPrinting file", filename)
print(data)
```

Sorry I tape so much of your time.
And thank you for the support!

laurensvalk, 2021-10-07:
There's a write function as well now, so

> Use the original firmware to create this file.

is technically no longer required. But it's still possible.

Our goal is that both Pybricks and the official firmware can see and access the same files.

So it would be interesting if you want to try creating a few files with both firmwares, and see if you can read them with the other firmware without trouble.

> Sorry I take so much of your time.

On the contrary, thanks for testing this undocumented experimental stuff :wink:

BertLindeman, 2021-10-07:
> You might have seen it on the REPL.
Should have, but did not look further than "no init"

I'll shut up now.

BertLindeman, 2021-10-07:
Good background, Thanks

laurensvalk, 2021-10-08:
> start the hub by pressing the left button and the middle button, so the runtime system is not started.

This works indeed. Alternatively, just press the red stop button in Thonny. This stops the runtime.

> Procedure worked

Excellent!

laurensvalk, 2021-11-22:
Closing since this has been answered.

closed 2021-11-22 by laurensvalk