pybricks/support #499: STATUS_BREAKPOINT running code.pybricks.com and beta.pybricks.com

BertLindeman, 2021-09-30:
**Describe the bug**
Trying to run a program results in a page that reports: error code  "STATUS_BREAKPOINT"

**To reproduce**
Steps to reproduce the behavior:
1. Go to https://code.pybricks.com/ on Microsoft Edge Versie 94.0.992.31 (Officiële build) (64-bits)
   Running on Win10
   Also code.pybricks.com "does" this
2. Pair with some hub (newest firmware or older firmware / on primehub, movehub or Technichub)
3. Start the program
4. See the error

**Expected behavior**
The python program should run...

**more info**
Restart of MS-Edge does not help.
If I remember correclty MS-Edge was updated today or yesterday.

**NO problem running Google Chrome Version 94.0.4606.61 (Official Build) (64-bit)**

**Screenshots**
[image]

**Reduced program**
``` python
from pybricks import version
print(version)
```

dlech, 2021-09-30:
If the problem happened because of an Edge update, the problem should be reported to Microsoft.

BertLindeman, 2021-09-30:
OK, David.

I understand your response. And right your are.
I chose to report the problem here, as I have not seen the problem on any other webpage I use.

In the meantime I have installed the current Chrome beta ```Versie 95.0.1020.9 (Officiële build) beta (64-bits)``` and there I did not (yet?) have the problem.
So I assume a next update will solve this.

Closing, due to above observations.

closed 2021-09-30 by BertLindeman