pybricks/support #501: [Bug] The "light.off()" command does not work with MoveHub

pavel-svoboda, 2021-10-04:
I am super-excited with the PyBricks, so cool - THANKS !!!

**Describe the bug**
I have created the following code with "light.off()" command. It works perfectly with the TechnicHub, but the same code does not work with the MoveHub. Switching the light on and other functions work perfectly.

**To reproduce**
Steps to reproduce the behavior:
1. Check the code below (the problem is with the last lines - see comments)
2. Get the Remote (LEGO Powered Up 88010 Remote Control)
3. Install code into MoveHub (LEGO Powered Up 88006), plug LED lights (LEGO Powered Up 88005) into port C
4. The Left red button switches the lights on, the Right red button should switch the light off, BUT it does nothing.

If I do the same with LEGO Powered UP 88012 TECHNIC Hub, the light works perfectly, so it is probably a problem specific to the MoveHub.

The code:
# Remote is currently in beta. This program only works with firmware
# installed from <https://beta.pybricks.com>.

from pybricks.pupdevices import Motor, Remote, Light
from pybricks.parameters import Port, Direction, Stop, Button
from pybricks.tools import wait

# Initialize the motors.
front_a = Motor(Port.A, Direction.COUNTERCLOCKWISE)
front_b = Motor(Port.B, Direction.COUNTERCLOCKWISE)

# Lower the acceleration so the car starts and stops realistically.
front_a.control.limits(acceleration=2000)
front_b.control.limits(acceleration=2000)

# Connect to the remote.
remote = Remote()

# Initialize the light.
light = Light(Port.C)

# Now we can start driving!
while True:
    # Check which buttons are pressed.
    pressed = remote.buttons.pressed()

    # Choose the drive speed based on the right controls.
    drive_speed_a = 0
    if Button.LEFT_PLUS in pressed:
        drive_speed_a += 1000
    if Button.LEFT_MINUS in pressed:
        drive_speed_a -= 1000

    drive_speed_b = 0
    if Button.RIGHT_PLUS in pressed:
        drive_speed_b -= 1000
    if Button.RIGHT_MINUS in pressed:
        drive_speed_b += 1000

    # Apply the selected speed.
    front_a.run(drive_speed_a)
    front_b.run(drive_speed_b)

    # Lights control - this is where the problem is
    if Button.LEFT in pressed:
        light.on(100)
    if Button.RIGHT in pressed:
        light.off()
    
    # Wait.
    wait(10)

BertLindeman, 2021-10-06:
Reduced the program to only use the light device.
The problem is seen on the movehub, not on Technic hub. 
I have not tested all combinations, but the problem is **not** in version `('movehub', '3.0.0', '8845360 on 2021-06-08')` 

The program I used:
``` python

# Source: Z:\home\bert\py\pybricks\issue\issue_#501_try2.py

# works on         ('technichub', '3.1.0b1', 'v3.1.0b1-8-ge82dd6a2 on 2021-09-28')
# and on           ('technichub', '3.1.0b1', 'v3.1.0b1 on 2021-09-21')
# does NOT work on ('movehub', '3.1.0b1', 'v3.1.0b1-14-g7ca4742d on 2021-09-29')
# and NOT       on ('movehub', '3.1.0b1', 'v3.1.0b1 on 2021-09-21')
# WORKS on         ('movehub', '3.0.0', '8845360 on 2021-06-08')

# plugged in the leds while hub was OFF.

from pybricks.pupdevices import Light
from pybricks.parameters import Port
from pybricks.tools import wait
from pybricks import version
print(version)

# Initialize the light.
light = Light(Port.C)

# Blink the light forever.
while True:
    # Turn the light on at 100% brightness.
    light.on(10)
    wait(500)

    # Turn the light off.
    light.off()
    wait(500)

```

There **was** a moment that I tought a beta version **was** working, but it never happened again, so it must be **me**,

BertLindeman, 2021-10-06:
Tried to find firmware files from continuous integration builds to find something between
`('movehub', '3.0.0', '8845360 on 2021-06-08')` and
`('movehub', '3.1.0a1', '3ab93ae on 2021-06-23')`
but they are expired as far as I saw.

On `('movehub', '3.0.0', '8845360 on 2021-06-08')`  ```light,off()``` works
and on `('movehub', '3.1.0a1', '3ab93ae on 2021-06-23')` it does not work.

closed 2021-10-15 by dlech

BertLindeman, 2021-10-16:
Confirmed fixed at C.I. build Build #1321 (obviously 😄 )
reported version:
```
('movehub', '3.1.0b1', 'v3.1.0b1-42-g06ca715b on 2021-10-15')
```
Thanks David