pybricks/support #509: [Question] How to change the direction of gyro sensing in EV3 on Pybricks?

STEM-Robotix, 2021-10-16:
**Question**
I can't change the direction of the gyro sensor on an EV3 in Pybricks to COUNTERCLOCKWISE.  

**Context**
It seems like this should be easy, The documentation is clear.  Seems like I should only have to add a second parameter when I create the gyro object. But when I go to run I get an error of an unexpected parameter.  Not sure what's wrong here?

gyro = GyroSensor(Port.S2, positive_direction=COUNTERCLOCKWISE)

I use the direction parameter for both motors I'm using with no errors as you can see.  Any help here would be much appreciated.  Thanks,

**Screenshots**
# Initialize the motors.
left_motor = Motor(Port.B, positive_direction=Direction.COUNTERCLOCKWISE)
right_motor = Motor(Port.C, positive_direction=Direction.COUNTERCLOCKWISE)
# Initialize the color sensor.
color_left = ColorSensor(Port.S1)
color_right = ColorSensor(Port.S4)
# Initialize the gyro sensor
gyro = GyroSensor(Port.S2, positive_direction=Direction.COUNTERCLOCKWISE)        ### This is Line 32 ###

Starting: brickrun --directory="/home/robot/Mazer" "/home/robot/Mazer/Mazer2.py"
----------
Traceback (most recent call last):
  File "/home/robot/Mazer/Mazer2.py", line 32, in <module>
TypeError: extra keyword arguments given
----------
Exited with error code 1.

laurensvalk, 2021-10-17:
Thanks for reporting this. I can confirm this issue.

This looks like a mistake in the documentation.

The correct argument is 'direction', instead of 'positive_direction'.

You could also leave out the name altogether.

STEM-Robotix, 2021-10-18:
Thanks so much, that fixed the issue.  Works great now.

closed 2023-03-20 by laurensvalk